Klipper3d/klipper #4250: BTT SKR Mini E3 EXP1 pinout aliases reversed

Asterchades, 2021-05-03:
The provided example configurations for the BTT SKR Mini E3 boards list the following information:

```
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>
```
_ref: https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-mini-e3-v2.0.cfg_

At the same time the provided LCD configuration file also contains the following:

```
######################################################################
# Plug pin locations
######################################################################

# Some micro-controller boards and displays use inconsistent labeling
# for the EXP1 and EXP2 headers. The following diagram shows the
# correct location of pin 1 along with ground and power pins on the
# EXP1 and EXP2 plugs:
#
#          EXP1:                        EXP2:
#   +-----------------+          +-----------------+
#   |  o  o  o  o  5V |          |  o  o  o  o  o  |
#   |  1  o  o  o GND |          |  1  o  o  o GND |
#   +------     ------+          +------     ------+
#
# Some boards may have the cutout in the wrong location. If so, it may
# be possible to carefully pry the plastic header off of the pins with
# a small screwdriver, then correct the orientation and reapply the
# plastic header.
```
_ref: https://github.com/KevinOConnor/klipper/blob/master/config/sample-lcd.cfg_

In the case of the BTT SKR Mini E3 series (and possibly others in the BTT SKR line-up, if not other boards as well) this is demonstrably incorrect, and the exact reverse of the physical hardware. On the physical hardware pin 1, as represented by the square pad, corresponds to +5V, per the silk screening on the rear of the board. Multimeter measurement can be used to confirm this, even when the [Display] section of the configuration is absent.

_ref: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-PIN.pdf_
[image]
_The image is not as clear as I would like (removing my board is difficult so I am not able to obtain a more dedicated image) but it is readable, and the clearly different pad is visible at +5V in the lower left._

Reversal of this order should not be harmful given the logic levels involved but will result in any device using this interface being rendered inoperable without corresponding alteration of the pin assignments or reversing the physical connection at one end.

Sineos, 2021-05-03:
https://github.com/KevinOConnor/klipper/blob/master/config/sample-lcd.cfg is a generic information that can potentially be applied to hundreds of board variants and also clearly states:

```
# Some boards may have the cutout in the wrong location. If so, it may
# be possible to carefully pry the plastic header off of the pins with
# a small screwdriver, then correct the orientation and reapply the
# plastic header.
```

This is exactly the reason for the `aliases` definition providing a convenient way to remap between hundreds of boards and hundreds of LCD variants.

Asterchades, 2021-05-03:
The header on these boards has the key in the "correct" place.

Am I to take away from this, then, that the EXP1_x designators are not what Klipper refers to the pins as internally, rather that they're a means to make some kind of "standard" out of the header? In which case, wouldn't that make the aliases completely superfluous? They already have standardised functions, even if not standardised pin numbers - one pin will always be VCC, the one directly opposite will be PB5, no matter what pin number they are. Changing those from VCC and PB5 to EXP1_10 and EXP1_1 respectively makes them no easier to allocate.

Or is this simply some kind of fail-safe in the event some board manufacturer decides to change the order of the pins on their EXP header/s, provided they don't do anything too difficult with VCC and GND?

closed 2021-05-03 by Asterchades