Klipper3d/klipper #4259: Troubleshooting not able to connect to SKR 1.3

disneytoy, 2021-05-04:
My issues:

Fresh Mainsail/Klipper install on a Pi Zero. All updates. (The SKR 1.3 is connected via the TFT pins no USB)

1) Klipper won't connect to the SKR 1.3 (via UART)

            Klippy-State: disconnected: Printer is not ready

2) I try to check the serial port via Putty and get:
         ls /dev/serial/by-id/*
         ls: cannot access '/dev/serial/by-id/*': No such file or directory

3) In printer.cfg I did set: serial: /dev/ttyAMA0

4) Made SKR firmware (LPC1768+100mhz+SMOOTHIEWARE switched USB to SERIAL)         
                It appears to have flashed correctly. Windows does not see the USB port (because it is turned off?) 

How can I tell if my serial UART connection is working?

In Nero3d video he mentioned setting it as UART2? But that was not an option when I used Rasp Config?

Any ideas how to troubleshoot this?

I followed Nero3d's video:

1. Swapping ports used by GPIO and Bluetooth sudo nano /boot/config.txt
      Move the cursor to the very end and add:dtoverlay=pi3-miniuart-bt 
 
2. Disabling the serial console sudo nano /boot/cmdline.txt
      Look for following string (text) and delete itconsole=serial0,115200 
 
3. Raspi-config stuff sudo raspi-config
     Go to 'Interfacing Options' Then P6 - Serial
     Then No Then Yes Then go down to finish and reboot. 
  
 4. Rebuild your Klipper MCU firmware, unselecting "Use USB for communication (instead serial). Flash updated firmware to your board 

 5. Update your printer.cfg:  serial: /dev/ttyAMA0

Camphul, 2021-05-04:
I had the same issue with my SKR 1.4 Turbo w/ uart connection. There seems to be a bug in the code regarding baudrates. Try lowering it from 250000 to 115200(worked for me). I will reflash mine once this issue is resolved.

KevinOConnor, 2021-05-04:
There is a known issue with serial connections to the lpc176x - #4194 

-Kevin

KevinOConnor, 2021-05-06:
I'm going to close this in favor of the existing #4194 .  (As indicated in that issue, the underlying problem should now be fixed.)

-Kevin

closed 2021-05-06 by KevinOConnor