Klipper3d/klipper #4264: docs: Use Markdown-lint for Checking Markdown Format

dingyifei, 2021-05-04:
Instead of checking for redundant whitespace, markdownlint can provide a standard and comprehensive method to standardize the doc styling. 

A link for anyone whos interested
https://github.com/DavidAnson/markdownlint-cli2

KevinOConnor, 2021-05-06:
Thanks - I'll try to take a look at that tool.

-Kevin

Sinclair5, 2021-06-03:
> Thanks - I'll try to take a look at that tool.
> 
> -Kevin

hello,   klipper is working great for me.  thanks for faking it free,   but i also wanted  to talk to you about how i can control an additional stepper motor that those my own personal thing.  pleas this  is really important  to me.  if you respond  ,  i can give you my email or anything so i can explain in more details.    please help me

Sinclair5, 2021-06-03:
or if you are in the discord server  i can just message u there

dingyifei, 2021-06-03:
> or if you are in the discord server  i can just message u there

you could try to discuss it in the Klipper Discord server first, usually a few devs are hanging out there regularly. Additionally, you could open a separate issue for discussion on the things you are working on.

Sinclair5, 2021-06-03:
i tried  but the dont really try to hlp  because the said it isnt fdm related. But
 i really need help from someone who contributed to the  code and has programming experience

Sinclair5, 2021-06-03:
> > or if you are in the discord server  i can just message u there
> 
> you could try to discuss it in the Klipper Discord server first, usually a few devs are hanging out there regularly. Additionally, you could open a separate issue for discussion on the things you are working on.

will you  be able to help?

dingyifei, 2021-06-04:
> will you  be able to help?

I wish I can help, but I'm an absolutely potato programmer. To get contributors, I think you should try to maintain your project regularly and build up a community. People would show up in ways I really don't know how to explain. I suggest you to read this book to get some insights about how open source community operates and how to grow your program, it's available in multiple languages.

[The Cathedral and the Bazaar](http://catb.org/~esr/writings/cathedral-bazaar/)

If you want to discuss the details about your project with other devs, you can post it in some forums or some IRC chat.

Sinclair5, 2021-06-04:
ok. thanks.  do u know of a way i can also contact kevin himself?

KevinOConnor, 2021-06-04:
@Sinclair5 - your messages are off-topic here - please do not post further here.  For contact info see: https://www.klipper3d.org/Contact.html

-Kevin

Sinclair5, 2021-06-04:
ok.  sorry

closed 2021-07-18 by github-actions[bot]