Klipper3d/klipper #4277: Rotation Distance is wrong in cfgprinter-twotrees-sapphire-plus-2020.cfg

Wilk86, 2021-05-09:
Sapphire Plus he's 2GT Pulley 20 Tooth, not 16.

microsteps: 16
rotation_distance: 40

and some have MBG clowns # gear_ratio: 3:1 # BMG gear ratio

i don't know how github works. pull request is what I had to do?

mninja79, 2021-05-09:
The microsteps are correct, they  are for the stepper motor drivers not the stepper motor. https://www.klipper3d.org/Rotation_Distance.html
The rotation_distance for the sapphire plus is 40 for x and y.
For the extruder if you have a clone dual drive and you add the gear ratio: 3:1 then the rotation distance is about 23.52.
If you donot include the gear ratio in the config then the rotation distance is around 7.77
But you will have to tune to get the correct rotation distance.

KevinOConnor, 2021-05-10:
@thucar , @pranas - do you have any comments on this ticket?

If it's a BMG extruder, the gear ratio should be 50:17 .  It would be great if someone could take a look at the actual extruder gears and confirm the number of teeth on them.

-Kevin

pranas, 2021-05-12:
I no longer run the stock extruder but it was a BMG clone so it would make sense that the gear ratio should be 50:17. 

Regarding `rotation_distance`, I still use the old step distance of `.0125`, and if it was converted according to the manual that should be 40. I think I had the value wrong in the original config. I'll open a PR right now.

mninja79, 2021-05-12:
@pranas  FYI I'm using the stock extruder  and used  the steps in the manual to get rotation distance using measure and trim.
Without gear ratio added to the config file . rotation_distance = 7.77
With the gear ratio added (50:17) to config file. rotation_distance = 23.52

pranas, 2021-05-12:
Thanks, @mninja79. I amended the extruder fix to my PR as well.

pranas, 2021-05-13:
@KevinOConnor I think this can now be closed since all the issues here were addressed with PR #4287

closed 2021-05-21 by github-actions[bot]