Klipper3d/klipper #4278: TMC driver settings unable to select negative number

TronskiFPV, 2021-05-09:
While trying new TMC2209 chopper settings I was unable to use -1 for HEND and received a error message in Fluidd that it must be at least 0

from the datasheet it looks like it can be set from -3 to 12
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2130_datasheet.pdf

Effectivity: TMC2209 and probably most other spreadcycle TMC chips

Impact: unable to fine tune stepper motor/driver combos. 

Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 157, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 141, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 130, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/tmc2209.py", line 94, in load_config_prefix
    return TMC2209(config)
  File "/home/pi/klipper/klippy/extras/tmc2209.py", line 80, in __init__
    set_config_field(config, "hend", 0)
  File "/home/pi/klipper/klippy/extras/tmc.py", line 64, in set_config_field
    val = config.getint(config_name, default, minval=0, maxval=maxval)
  File "/home/pi/klipper/klippy/configfile.py", line 62, in getint
    minval, maxval, note_valid=note_valid)
  File "/home/pi/klipper/klippy/configfile.py", line 45, in _get_wrapper
    % (option, self.section, minval))
Error: Option 'driver_HEND' in section 'tmc2209 stepper_x' must have minimum of 0
webhooks client 1970859224: New connection
webhooks client 1970859224: Client info {'program': 'Moonraker', 'version': 'v0.6.0-22-gc53b95a'}
[klippy.log](https://github.com/KevinOConnor/klipper/files/6448145/klippy.log)

meteyou, 2021-05-09:
this is because klipper writes the register value directly and there are no negative numbers. therefore -1 == 2.

-3 => 0
-2 => 1
-1 => 2
0 => 3
1 => 4
...
12 => 15

TronskiFPV, 2021-05-09:
Thank you so much! Man this gets tricky with Klipper. I wish it were written to be more comparable with Marlin, but I get it...

I will try to add that to the documentation.

meteyou, 2021-05-09:
I'm pretty sure about all the values that start with "drivers_" in your config

TronskiFPV, 2021-05-10:
It looks like I either don't understand the driver manual or the numbers are different for each driver section. 

I tried to use a Marlin (3,-1,5) and that would be this in klipper using that no neg number format
driver_TOFF: 6
driver_HEND: 2
driver_HSTRT: 8

it wouldn't take the HSRT 8 even though its "5" and the setting can go to 8 the best I can figure. I'll hold off on updating the documentation until I can figure it out. 

this whole thing is to get a LDO-42STH40-1684MAC .9 deg motor to work well for a A/B motor on a coreXY with TMC2209 drivers. I don't have a oscilloscope, but it doesn't sound very happy with everything I've tried so far

TronskiFPV, 2021-05-11:
re-opening. I'll try to work on this soon

KevinOConnor, 2021-05-11:
This looks like a request for help, which we don't use github for.

That said, the behaviour above is documented at: https://www.klipper3d.org/TMC_Drivers.html#configuring-driver_xxx-settings

-Kevin

TronskiFPV, 2021-05-11:
yeah that documentation needs work so I re-opened it. No worries, I have my drivers working myself now anyhow.

closed 2021-05-12 by github-actions[bot]