Klipper3d/klipper #4290: Unhandled exception during print Palette2

BenWhite27, 2021-05-13:
Hi
Just got an unhandled exception during my first print with a Palette2 device.  I've attached the klippy.log file and pasted what I think is the interesting part below.
[klippy.log](https://github.com/KevinOConnor/klipper/files/6474874/klippy.log)
```
Unhandled exception during run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 201, in run
    self.reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/extras/palette2.py", line 576, in _run_Read
    self.p2cmd(text_line)
  File "/home/pi/klipper/klippy/extras/palette2.py", line 519, in p2cmd
    func(params)
  File "/home/pi/klipper/klippy/extras/palette2.py", line 351, in p2cmd_O20
    splice = self.omega_splices[self.omega_splices_counter]
IndexError: list index out of range
Transition to shutdown state: Unhandled exception during run
```

The print had almost finished the first layer of the wipe tower, I've attached the gcode it was working on as well which was sliced in SuperSlicer 2.3.56 (Nightly) and post processed with P2PP v7.01.11 (latest).  I was using a Palette 2S Pro.
[MosaicPokerChip_Clear.mcf.zip](https://github.com/KevinOConnor/klipper/files/6474922/MosaicPokerChip_Clear.mcf.zip)

@cmroche: Hope you don't mind me pinging you as this might land in your court.  I'm in the Discord server as `@wubba` if you need to speak to me directly and don't believe this to be a bug.

Regards
Ben

cmroche, 2021-05-13:
Hey @ben-kaye thanks for the info, however this is a bit perplexing. Your GCode file has only 18 pings, both the `O26` line and `O30` lines correctly show this.

But your palette asked for data about a non-existant 19th ping, which caused this error.

Do you have this with other prints, does it happen if you retry the print?

BenWhite27, 2021-05-13:
Hey @cmroche 
Thanks for looking at this so quickly.  This was my very first print with the Palette, it took quite a bit of work getting SuperSlicer to work with P2PP.  It's a bit late to retry it tonight but I will try and squeeze in another attempt after work tomorrow.
If it helps the slicer reported 17 tool changes which sounds correct in my head and I remember the Palette reporting ??/18 splices on the screen during the print.  It's the latest available firmware on the Palette as well.

Will report back tomorrow or at the weekend if I have any luck.

BenWhite27, 2021-05-15:
@cmroche I tweaked a few filament settings in the slicer as I had some under extrusion on the first print, and re-printed without issues. 

I'll compare the gcode later today and see if I can spot anything that could have caused this.

Let's hope this was a one off or a silly mistake on my part.

cmroche, 2021-05-15:
@BenWhite27  When I looked more closely at the log file you attached, I noticed that there was an instance of the Palette sending unexpected data earlier on, also leading to an error. Your USB cable might be part of the issue, if you can, perhaps try replacing it and see if reliability increases.

BenWhite27, 2021-05-15:
> Your USB cable might be part of the issue, if you can, perhaps try replacing it and see if reliability increases.
Will do, it's far too long anyway.

The print worked the 2nd time so I'm fairly sure there's no bug here and you had a much longer successful print when you first got this working - My thanks for that!!

I'll close off this issue and continue to diagnose this end.

closed 2021-05-15 by BenWhite27