Klipper3d/klipper #4292: Bed probing returns a flipped Y axis after Klipper update

vashqiri, 2021-05-14:
After updating to Klipper latest version (0.9.1-499), it appears that the Y axis seems to be flipped once the values are sent back to BedMeshVisualizer on Octoprint.
This was confirmed by flashing Marlin back and performing a bed probing without touching the bed.
The bed's tilt has been voluntarily exaggerated for testing purposes and confirming the issue.

Bed mesh on latest Klipper :
[image]

Bed mesh on Marlin :
[image]

Arksine, 2021-05-15:
The way Bed Mesh reports values hasn't changed since the module was first introduced..   I believe that visualizers have options to invert the Y axis if necessary.

vashqiri, 2021-05-15:
> The way Bed Mesh reports values hasn't changed since the module was first introduced.. I believe that visualizers have options to invert the Y axis if necessary.

Yes, they do. It displays the mesh as it should be after activating the option (which I did to be able to properly level). No big deal, it might just be a little counter intuitive to some people.

The issue can be closed if it's not seen as something to be changed.

Arksine, 2021-05-15:
> No big deal, it might just be a little counter intuitive to some people.

I believe that at one point Marlin to reported the mesh inverted, as does Prusa Firmware.  It is done this way so that users who want to look at the output are doing so as if they are looking down on the bed, thus YMax is reported first.  I don't intend to change the way its reported unless Kevin requests that I do so.

That said, `[bed_mesh]` also includes the `BED_MESH_MAP` gcode which reports the mesh in JSON.  These visualizers could support that gcode if they want to avoid having users configure the visualizer for an inverted y-axis.

vashqiri, 2021-05-15:
> I believe that at one point Marlin to reported the mesh inverted, as does Prusa Firmware. It is done this way so that users who want to look at the output are doing so as if they are looking down on [...]
Interesting. :)
On cartesian printers with a Z-fixed bed, bottom left point on the mesh should be XMin/YMin when looking from above (which is mostly the case), not Xmin/Ymax. Unless you look at the bed from below.

> That said, `[bed_mesh]` also includes the `BED_MESH_MAP` gcode which reports the mesh in JSON. These visualizers could support that gcode if they want to avoid having users configure the visualizer for an inverted y-axis.
Maybe BedLevelVisualizer supports this. Gotta try this.

Arksine, 2021-05-15:
> On cartesian printers with a Z-fixed bed, bottom left point on the mesh should be XMin/YMin when looking from above (which is mostly the case), not Xmin/Ymax. Unless you look at the bed from below.

Correct.  If you look at the mesh data from `BED_MESH_OUTPUT`, the, it will be something like this 
```
Measured points:
-0.325000 -0.293056 -0.239333 -0.202500 -0.169204 -0.121796 -0.092500 -0.079000 -0.064611 -0.060000 -0.082889 -0.123222 -0.150000 -0.165481 -0.186741 -0.205000 -0.248426 -0.316519 -0.357500
-0.290296 -0.258435 -0.204971 -0.168778 -0.138703 -0.096702 -0.071315 -0.062996 -0.056160 -0.055287 -0.073253 -0.104388 -0.125352 -0.139510 -0.159697 -0.176778 -0.215530 -0.276145 -0.312611
-0.232481 -0.200700 -0.147569 -0.112389 -0.087704 -0.054761 -0.035907 -0.036135 -0.041737 -0.047019 -0.056878 -0.072865 -0.084259 -0.096261 -0.114720 -0.129889 -0.160978 -0.209328 -0.238389
-0.195000 -0.163037 -0.109741 -0.075000 -0.053907 -0.027037 -0.012500 -0.017926 -0.031019 -0.040000 -0.044981 -0.051852 -0.057500 -0.068296 -0.085870 -0.100000 -0.126611 -0.167778 -0.192500
-0.174352 -0.140719 -0.084765 -0.048796 -0.030239 -0.007919 0.003759 -0.002454 -0.016197 -0.025458 -0.030053 -0.036201 -0.041500 -0.052945 -0.071690 -0.086574 -0.113515 -0.155062 -0.180000
-0.148370 -0.112008 -0.051686 -0.013593 0.001488 0.017503 0.025352 0.019125 0.006201 -0.002583 -0.007764 -0.015006 -0.021167 -0.033939 -0.054771 -0.071148 -0.099950 -0.144253 -0.170833
-0.132500 -0.094574 -0.031759 0.007500 0.020250 0.032111 0.037500 0.031690 0.019991 0.011875 0.005898 -0.002829 -0.010000 -0.023444 -0.045250 -0.062500 -0.093519 -0.141315 -0.170000
-0.123667 -0.086164 -0.024129 0.014333 0.024590 0.032552 0.035926 0.032417 0.024950 0.019056 0.010478 -0.003053 -0.013028 -0.025221 -0.044575 -0.061333 -0.099709 -0.159901 -0.196130
-0.113611 -0.077219 -0.017123 0.019722 0.026505 0.029173 0.029907 0.030168 0.029487 0.027153 0.014469 -0.006662 -0.021056 -0.031140 -0.046411 -0.062278 -0.112371 -0.192332 -0.240593
-0.110000 -0.074444 -0.015778 0.020000 0.025148 0.025407 0.025000 0.027352 0.030370 0.030000 0.015306 -0.009611 -0.026250 -0.035537 -0.049213 -0.065000 -0.121648 -0.212574 -0.267500
-0.123074 -0.088481 -0.031305 0.003963 0.011891 0.016872 0.019148 0.020225 0.020605 0.019023 0.007546 -0.011864 -0.025630 -0.038289 -0.058142 -0.077259 -0.130848 -0.215910 -0.267204
-0.146926 -0.113784 -0.058821 -0.024185 -0.011189 0.002323 0.009407 0.007936 0.003145 -0.000870 -0.006561 -0.015996 -0.024509 -0.043012 -0.073582 -0.098296 -0.145263 -0.217998 -0.261685
-0.165000 -0.132778 -0.079222 -0.045000 -0.028880 -0.010093 0.000000 -0.002773 -0.010324 -0.015625 -0.018046 -0.021884 -0.027500 -0.049648 -0.086824 -0.115000 -0.158019 -0.223370 -0.262500
-0.189037 -0.157213 -0.104300 -0.070407 -0.053983 -0.034558 -0.023870 -0.025093 -0.030238 -0.034306 -0.038070 -0.044476 -0.051778 -0.074723 -0.112896 -0.141796 -0.185976 -0.253081 -0.293259
-0.226630 -0.195306 -0.143216 -0.109815 -0.093547 -0.074161 -0.063074 -0.061175 -0.061314 -0.062847 -0.069844 -0.082271 -0.093333 -0.116857 -0.155249 -0.184593 -0.231539 -0.303128 -0.346019
-0.255000 -0.224185 -0.172926 -0.140000 -0.123722 -0.104111 -0.092500 -0.087722 -0.083278 -0.082500 -0.092481 -0.110463 -0.125000 -0.149130 -0.187815 -0.217500 -0.266148 -0.340463 -0.385000
-0.299463 -0.270284 -0.221660 -0.190093 -0.172617 -0.150449 -0.136352 -0.124564 -0.109233 -0.102981 -0.119947 -0.150986 -0.173796 -0.200329 -0.241161 -0.271926 -0.320340 -0.393910 -0.437963
-0.367037 -0.340464 -0.296041 -0.266630 -0.247190 -0.220859 -0.202759 -0.179896 -0.147380 -0.132491 -0.160490 -0.212155 -0.248037 -0.278407 -0.322680 -0.355130 -0.402799 -0.474572 -0.517481
-0.407500 -0.382500 -0.340611 -0.312500 -0.291870 -0.263019 -0.242500 -0.212963 -0.170093 -0.150000 -0.184648 -0.248741 -0.292500 -0.325185 -0.371537 -0.405000 -0.452185 -0.522815 -0.565000
```

The top left coordinate is (X-Min, Y-Max), the bottom left coordinate is (X-Min, Y-Min).  By default the visualizer sees this as inverted, presumably because it expects the first top line to start at (X-Min, Y-Min).

vashqiri, 2021-05-15:
Makes sense.
Unfortunately, the current visualizer can't read JSON output from BED_MESH_MAP (for now). I'll keep the checkbox inverting the Y axis on, thus closing the issue.

KevinOConnor, 2021-05-17:
Okay, this seems like a general topic of discussion and a github issue is probably not the best place for it.

-Kevin

closed 2021-05-17 by vashqiri