Klipper3d/klipper #4297: Bed heating PID is affected by nozzle heating

muhammadelmogy, 2021-05-16:
I have a problem that I can't understand why it is happening nor able to solve!

If I am heating the nozzle and the bed to a target temperature and If the bed reached its target temperature while the nozzle is still heating up there is nothing until now wrong but as soon as the nozzle reaches its target temperature, the bed temperature suddenly jump like 3 or 4 degrees up without any reason and stay raised for too much time like 1 minutes or so and start gradually decrease until it reaches its target with in other 3 or 4 minutes.

One thing I noticed that when this happens, the SSR(Solid state relay) does not really work at that point (when the temp. jumps)
but it keep trying to work later if the temp. start decreasing which makes it goes up again and this happens for the next 3-4 minutes until it gradually become stable 
One thing I am doing to avoid this is to start heating the nozzle and wait until the temperature become stable and start heating the bed after that.

Of course because of that behavior the print job does not start until bed temperature is stable
I tried to make PID tuning for both the bed and the nozzle several times before but is does not help
I though it might be the Arduino 5V source is not stable and have noise so the ADC does not have a clean reference, so I changed the power supply to better one but I still have the same issue 

I just recently moved from marlin and I was using it with my printer for several years and never had such an issue before.

I am using 220v 750W silicone heater for 300X300X4 mm bed and 60w custom nozzle.

[klippy.log](https://github.com/KevinOConnor/klipper/files/6489299/klippy.log)

JustAnother1, 2021-05-16:
A sudden jump in temperature especially on the bed can not happen in reality. So there must be either something wrong with your temp sensor, or with the software. That the SSR is also misbehaving seems to indicate that something is wrong with the way you have connected the two heaters and two temperature sensors.
That the temperature slowly goes back to normal after the jump could also be a capacitor charging /discharging due to a wiring issue or a software integration issue. Does it only happen with PID? Did you try Bang-Bang on the Bed?
Could you provide the schematic of how your heaters and temp. sensors are connected?

muhammadelmogy, 2021-05-16:
I have been using my current configuration for wiring with marlin for several years and I have never had this issue before.
This jump only happens when the nozzle reaches its target temp. even if the bed has already reached its target temp. for a while!
Some kind of offset added to the bed temp. reading when nozzle reaches its target temp. That could be the closest description for what I am seeing!
I never uses bang bang, also I tried increasing Pwm freq. put that didn't help

JustAnother1, 2021-05-17:
I just thought that it might be a strange effect caused ba the PID going wrong. Using Bang-Bang would be an option to rule that out.  Just trying to be helpful.

muhammadelmogy, 2021-05-17:
> I just thought that it might be a strange effect caused ba the PID going wrong. Using Bang-Bang would be an option to rule that out. Just trying to be helpful.

I appreciate it, thanks
I want to use PID since it was working fine with Marlin
Also this issue must be resolved since it could be some bug in the code!

muhammadelmogy, 2021-05-17:
I am sorry but I think this may be a bug not a request for help!
May be I used incorrect words so it appears as a request but in the matter of fact, I meant a bug!
What makes me think it is a bug is my printer was already working fine with marlin without experiencing this issue and after using Klipper I faced this problem, so basically this shouldn't be a hardware issue!

I think you need to define a clear guide lines to help people differentiate between reporting a bug and requesting for help!

muhammadelmogy, 2021-05-17:
I opened a thread on the forum and will discuss it there first to confirm if it was a bug or another issue
I will be closing this for now!

closed 2021-05-17 by muhammadelmogy