Klipper3d/klipper #4299: Macro frustration: weird bheaviours of the macro engine

kolergy, 2021-05-16:
Issue One:
The parser of the gcode macro is parsing the comments ->  I had comments where I expressed the unit in the comments as  mm³ parser did not like that (Modern system should parse unicode ) 

Issue Two:
Behavior is not consistent in previous comments there are mm³ but parser is happy about it

Issue Three:
It is even trying to execute comments : In the process of understand the weird logic that all parameters shall be a string I had the following line under comment:  ' ;{% set vol             = 1000 * MF / DE                                   %} ; number of mm^3 to extrude'
but on the execution of the macro it still try to execute it and I got the following error as it was clearly under comment:
'  | Error evaluating 'gcode_macro CAL_MF:gcode': TypeError: unsupported operand type(s) for /: 'str' and 'str''
I had to remove the line to eliminate the error. 
Are the comment really working ?

Issue Four:
No indication of line number in the error message! This help debugging... 
- when parsing the print.cfg
- when executing the macro

Issue Four:
And to finish my killer one it seems to execute the macro but not really & get stuck, it dose not wait for head to heat up even though it is a M109  and nothing happens except that the head go to the desired temp.
And there I have no clue of what is going on...

[klippy.log](https://github.com/KevinOConnor/klipper/files/6489482/klippy.log)

macro:
'# calibrate filament mass flow vs extrusion rate & temp
[gcode_macro CAL_MF]
default_parameter_TE: 200    ; Print temperature (°C)
default_parameter_MF:   1.0  ; Filament mass (g)
default_parameter_VF:   3.0  ; Filament volume flow (mm³/s)
default_parameter_DE:   1.24 ; Filament Density (g/cm³)
gcode:
    {% set extruder_config = printer.configfile.settings['extruder']    %}

    {% set MX_Ext_STD      = extruder_config.max_extrude_only_distance|float  %}
    {% set diam_fil        = extruder_config.filament_diameter|float          %}
    {% set volPmm          = diam_fil * diam_fil * 3.1416 / 4.0               %} ; filament volume per mm mm^3/mm of filament
    {% set vol             = 1000 * MF|float / DE|float                       %} ; number of mm^3 to extrude
    {% set len             = vol / volPmm                                     %} ; length to be extruded
    {% set speed           = VF|float  / volPmm                               %} ; extrusion speed in (mm/s)
    {% set len             = vol / volPmm                                     %} ; length to be extruded
    {% set dl              = (len / MX_Ext_STD)|abs()|int                     %} ; number of max lenght extrude rounds to perform
    {% set rl              = len - dl * MX_Ext_STD                            %} ; remaining length to extrude
    {% set et              = len / speed                                      %} ; extrusion time
        
    { action_respond_info("               Density of filament: " ~ DE     ~ " g/cm^3" ) }
    { action_respond_info("         Volume per mm of filament: " ~ volPmm ~ " mm^3"   ) } 
    { action_respond_info("          Temperature of extrusion: " ~ TE     ~ " C"      ) }
    { action_respond_info("          Volume Flow of extrusion: " ~ VF     ~ " mm^3/s" ) }
    { action_respond_info("   Mass of filament to be extruded: " ~ MF     ~ " g"      ) }
    { action_respond_info(" Volume of filament to be extruded: " ~ vol    ~ " mm^3"   ) }
    { action_respond_info("                Speed of extrusion: " ~ speed  ~ " mm/s"   ) }
    { action_respond_info("  Lengt of filament to be extruded: " ~ len    ~ " mm"     ) }
    { action_respond_info("            Max Lengt in one round: " ~ MX_Ext_STD ~ " mm" ) }
    { action_respond_info("  number of max extrude to perform: " ~ dl     ~ " (-)"    ) }
    { action_respond_info(" remaining filament to be extruded: " ~ rl     ~ " mm"     ) }
    { action_respond_info("            time for the extrusion: " ~ et     ~ " s"      ) }

    ; Part inspired from CNC Kitchen https://www.cnckitchen.com/blog/flow-rate-benchmarking-of-a-hotend
    ; Adapted to take variables & define a mass to be extruded
    M109 S{ TE }                                  ; Set Hotend Temperature to print temp 
    { action_respond_info("Priming th nozzle" ) }
    M83                                           ; relative extrusions 
    G1 F60                                        ; 60mm/min speed 
    G1 E20                                        ; prime with 20mm 
    { action_respond_info("Collect primming extrusion " ) }
    G1 F3                                         ; slow extrude for pause 
    G1 E0.1                                       ; 0.1mm extrusion 
    { action_respond_info("Measurement Extrusion " ) }
    G1 F{speed|round(0)|int}                      ; Xmm/min speed  
    {% for i in range(dl) %}
        { action_respond_info("In Loop: " ~ i ) }
        G1 E{MX_Ext_STD}                          ; Xmm extrusions trick max extrusion limit
    {% endfor %}
    G1 E{rl|round(0)|int}                         ; 
    { action_respond_info("Measurement Completed " ) }
    M104 S0                                       ; Turn-off hotend
   '
Console output:

17:38:28 | Measurement Completed
-- | --
17:38:28 | In Loop: 5
17:38:28 | In Loop: 4
17:38:28 | In Loop: 3
17:38:28 | In Loop: 2
17:38:28 | In Loop: 1
17:38:27 | In Loop: 0
17:38:27 | Measurement Extrusion
17:38:27 | Collect primming extrusion
17:38:27 | Priming th nozzle
17:38:27 | time for the extrusion: 268.817204301 s
17:38:27 | remaining filament to be extruded: 35.2828353796 mm
17:38:27 | number of max extrude to perform: 6 (-)
17:38:27 | Max Lengt in one round: 50.0 mm
17:38:27 | Lengt of filament to be extruded: 335.28283538 mm
17:38:26 | Speed of extrusion: 1.24725214761 mm/s
17:38:26 | Volume of filament to be extruded: 806.451612903 mm^3
17:38:26 | Mass of filament to be extruded: 1.0 g
17:38:26 | Volume Flow of extrusion: 3.0 mm^3/s
17:38:26 | Temperature of extrusion: 200 C
17:38:26 | Volume per mm of filament: 2.4052875 mm^3
17:38:26 | Density of filament: 1.24 g/cm^3
17:38:23 | CAL_MF
17:36:45 | Klipper state: Ready

zellneralex, 2021-05-17:
Will not answer all your questions but maybe a good start https://klipper.discourse.group/t/macro-creation-tutorial/30

FHeilmann, 2021-05-17:
I'll try to shed some light on where your issues come from and whether fixing them is in the scope of klipper:

Issues One, Two and Three, Four:

These are issues with the underlying templating engine klipper uses (`jinja2`) and possibly the library used to parse the config (pythons `ConfigParser`). As such, while I agree that these issues are annoying, they would have to be fixed in those two libraries, instead of klipper. In particular issue 4 is tricky here, since the error is generated when jinja2 attempts to parse the templated macro, and not when the config is parsed. Moreover, since jinja only sees the macro and not the entire config, line numbers might be useless, in particular since during macro template expansion, stuff such as loops are unrolled and line numbers change as a result

Issue Five: 
`{% %}` delimits blocks, `{}` delimits inline template statements. This is similar to how comments work in, e.g., C++ where block comments are delimited by `/* */` while line comments simply start with `//`. Notice that `{% %} is always used in a block, where a statement such as `for` or `if` is always accompanied by a matching `endfor` or `endif`

Issue Six:
Do your shared log files include such an instance? The logs you shared contain config errors (where the parsing of the commands fails) and issues related to your serial port where klipper seems to fail to send data to your boards. If you have an example of a log where issue 6 occurred, feel free to share it so we can take a look

Issue Seven:
This one is very tricky to explain. For starters, here is the lifecycle of a macro, when it is executed:

1) User (or gcode file) sends macro command
2) Klipper uses Jinja2 to expand any templating code inside the macro
3) Klipper queues up the gcode generated by the expansion step
4) At some point the expanded gcode reaches the front of the gcode queue and gets executed.

`action_` calls are executed during the expansion step (step 2 above), not when the gcode reaches the front of the queue. This is the reason why all your print statements happen before your normal gcode even begins execution. This is an unfortunate limitation but there is a workaround: Write your own `DEBUG` macro that is passed the string you actually want to debug, which then just calls `M400` followed by `action_respond_info` with the passed message. What that does is it delays the expansion (and thus display) of the message until the gcode queue reaches your macro, which results in the behavior you're looking for.

Hope that helps
-Florian

KevinOConnor, 2021-05-17:
I think it would be best to move this discussion to the Klipper Discourse - as github is not a good location for a general discussions like this.

-Kevin

kolergy, 2021-05-17:
@KevinOConnor You have written an excellent firmware enabling us to drive our machines and extract most of their potential. It is often the case that developers of good software want to ensure that people can use their software with a minimum of ease this is why this was neither a request for help nor a general discussion. There is a remaining of six issues that generates surprising behavior from the point of view of a user discovering & learning Klipper and therefore make writing macros for Klipper an awkward sport.

Even if it will sadden me, it could be very understandable that you are not intending to fix them in the short term as from the explanation of FHeilmann some are deeply rooted in architectural choices and might not be very easy to treat. However these are still issues that will adversely affect your users.

@FHeilmann Thanks a lot for clarifying the origin of some of the issues & providing workarounds. Concerning issue 6, sorry the logs where the full ones so you can see my full trials & errors to understand how this works I tried to reproduce it but could not so I guess I was confused by something else

Here is a log of the working run of the macro which starts at line 316
[klippy-Working.log](https://github.com/KevinOConnor/klipper/files/6497393/klippy-Working.log)

For info before my initial post I did read the following pages that I had to chase on different repositories & servers:
- https://klipper.discourse.group/t/macro-creation-tutorial/30
- https://jinja.palletsprojects.com/en/2.10.x/
- https://github.com/Desuuuu/klipper-macros
- https://github.com/KevinOConnor/klipper/blob/master/docs/Command_Templates.md
- https://github.com/KevinOConnor/klipper/blob/master/docs/Status_Reference.md
- https://www.klipper3d.org/Config_Reference.html
- https://www.klipper3d.org/Command_Templates.html
- https://www.klipper3d.org/G-Codes.html

KevinOConnor, 2021-05-18:
> There is a remaining of six issues that generates surprising behavior from the point of view of a user discovering & learning Klipper and therefore make writing macros for Klipper an awkward sport.

I understand, but we only use github to share work that has been done to improve klipper.  If you're actively working on the above then we can keep this ticket open to track your progress.  Otherwise, we use Klipper Discourse (or Discord) for discussions.

-Kevin

kolergy, 2021-05-18:
I do not think I have the competences to sort those problems but I can try to work on the documentation if that can help.

closed 2021-05-20 by github-actions[bot]