Klipper3d/klipper #4314: Unable to use adxl345 on RPi4b

drdeimos, 2021-05-21:
OS: Ubuntu 20.04.2 LTS (aarch64)
Hardware: Raspberry Pi 4 Model B Rev 1.4

Steps to reproduce:
1) config:
```
[adxl345]
cs_pin: rpi:gpio8
```
2) run as root:  `# /usr/local/bin/klipper_mcu` 
3) Run klippy: ` /home/klipper/klippy-env/bin/python /home/klipper/klipper/klippy/klippy.py -a /tmp/klippy_uds /home/klipper/klipper-cfg/printer.cfg -l /tmp/klippy.log`
4) Invoke `FIRMWARE RESTART`

Got from klippy:
>error: MCU 'rpi' error during config: Unable to open out GPIO chip line
From `klipper_mcu`:
>Got error -1 in gpio_out_reset get line: (16)Device or resource busy

log, config, gpioinfo in attach

[klippy.log](https://github.com/KevinOConnor/klipper/files/6524511/klippy.log)
[printer.conf.txt](https://github.com/KevinOConnor/klipper/files/6524512/printer.conf.txt)
[gpioinfo.txt](https://github.com/KevinOConnor/klipper/files/6524519/gpioinfo.txt)

popshansen, 2021-05-22:
It works with Raspberry Pi OS/Octopi. I have used the ADXL345 on my Raspberry Pi 4b with no problems.

drdeimos, 2021-05-23:
Hmm... Test same config on octopi and get same result.

popshansen, 2021-05-23:
Have you tried the setup described in https://github.com/KevinOConnor/klipper/blob/master/docs/Measuring_Resonances.md ?

You have cs_pin: rpi:gpio8 but it should be cs_pin: rpi:None and enabled the SPI in raspi-config.

drdeimos, 2021-05-23:
I read it, but it is not obvious that when choosing "rpi:None", the klipper uses spi devices initialized by the raspberry itself.
And what kind of select chip is used?
What if I connect a second accelerometer, how can I split them along the axes?

drdeimos, 2021-05-23:
P.S. After change to `rpi:None` i got:
>!! Invalid adxl345 id (got 0 vs e5)

from `ACCELEROMETER_QUERY`

popshansen, 2021-05-23:
In what setup is it required to have two adxl345 ?
I assume it is input shaper you are tying setup, in most cases you measure one axis at the time.

drdeimos, 2021-05-23:
I don't want to carry or switch the accelerometer between the two axes.
Two fixed accelerometers: to the carriage and to the bed - two spi interface.

popshansen, 2021-05-23:
I have no used two adxl345, but you properly need to give your adxl345 names (like [adxl345 axisX] / [adxl345 axisY] ) and set the spi_bus: for the each adxl345 it is connected to. Have a look in the Config_Reference.md under adxl345 section on options.

drdeimos, 2021-05-24:
Update:
Original issue - resolved. Cause - defective accelerometer board.

The question remains: How to use the second SPI line on the same RPi4

popshansen, 2021-05-24:
> The question remains: How to use the second SPI line on the same RPi4

As I wrote in my previous answer you need to put each adxl345 on it's own spi bus. This link describe where you can find the GPIO's for each SPI busses on the RPI 4b (BCM2711): https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md
Then use the spi_bus parameter like this spi_bus: spidev2.0
For RPI you have the following SPI busses to choose from: spidev0.0, spidev1.0, spidev2.0 and spidev3.0

drdeimos, 2021-05-24:
Awsome. Thank you.

closed 2021-05-26 by github-actions[bot]

ddacunha, 2021-07-19:
Wouldn’t chaining the accelerometers in [Independent Slave Configuration](https://en.wikipedia.org/wiki/Serial_Peripheral_Interface#Independent_slave_configuration) and specifying the CS GPIO on the Raspberry Pi work? My understanding is that the X and Y tests are not running at the same time.

madorax251, 2021-08-07:
you can check at klipper.wiki