Klipper3d/klipper #4324: "Timer to Close" mcu shutdown

Stephan3, 2021-05-26:
On one of my pritners i run a raspi4 with klipper. On that printer it happened now the second time that it shuts down with "Timer to Close". I ran tha same 22h plate the 2 days before just fine but now the third stopped after some time. Its the same gcode file, just pushed "play" no upload, no nothing.
See the log attached. 
[klippy.zip](https://github.com/KevinOConnor/klipper/files/6545245/klippy.zip)

KevinOConnor, 2021-05-26:
We don't use github to help diagnose issues like this.  A "timer too close" is typically the result of host cpu or connectivity issues.  If you want to track it down, then the load graph and log analysis tools (see docs/Debugging.md) can be helpful.  Otherwise, best to report on Klipper Discord or Discourse to see if others can help.

-Kevin

Stephan3, 2021-05-26:
Thanks for your answer. I allready did what you said. The Graph looks not Horroble.
At 1 a 22h print(same gcode) ended just successfully. at 2 it died with timer to close. 

[image]

KevinOConnor, 2021-05-27:
FYI, you'll want to look at the system stats, not the mcu stats - run `graphstats.py --help` for details.

-Kevin

closed 2021-05-29 by github-actions[bot]

Stephan3, 2021-05-31:
I replaced now the PI, the Cable, reinstalled everything and it happended again. 3 days it printed fine on day 4 it died with timer to close. See the graphs i created. Wether i am blind or ther is nothing. I dont know what i can do next.

[image]
[image]