Klipper3d/klipper #4348: Failed to home probe: Timeout during homing after latest klipper (538)

cfelicio, 2021-06-05:
1 - I have made no mechanical or configuration changes on the printer whatsoever
2 - I updated from version 4xx to the latest just now (538)
3 - Now, when I try to run BED_MESH_CALIBRATE, I get "Failed to home probe: Timeout during homing"
4 - Homing the Z with the probe still works fine

Attaching log and configuration file. Thanks! I'll just revert to an older release for now.
[klippy.log](https://github.com/KevinOConnor/klipper/files/6601982/klippy.log)
[config.zip](https://github.com/KevinOConnor/klipper/files/6601983/config.zip)

cfelicio, 2021-06-05:
Other commands like probe_accuracy also seem to work fine

[image]

cfelicio, 2021-06-05:
Just to be sure, also re-ran initial tests per documentation, and everything seems fine with the probe:

[image]

KevinOConnor, 2021-06-05:
Sorry, we don't use github to help diagnose problems with individual printers.  You'll want to bring this up on the Klipper Discord or Discourse servers.  (That aside, you likely need to add `position_min: -1` to your stepper_z config section.)

-Kevin

cfelicio, 2021-06-05:
Thanks Kevin, I guess I stayed too far behind on versions, as everything was working fine before I upgraded. I'll go ahead and read the documentation, as well as closing this one. Cheers!

closed 2021-06-05 by cfelicio

cfelicio, 2021-06-05:
Umm, something really odd going on lol, now I'm able to calibrate, but the first measurement indicates the bet is really off :-)

[image]

I'll move this one to Discord.

Cheers!

Talha909, 2021-06-07:
> 1 - I have made no mechanical or configuration changes on the printer whatsoever
> 2 - I updated from version 4xx to the latest just now (538)
> 3 - Now, when I try to run BED_MESH_CALIBRATE, I get [...]

did you find any solution. I got the same problem.

cfelicio, 2021-06-07:
> did you find any solution. I got the same problem.

The problem fixed itself after I did a clean install and reflash. Worked with the latest version too. You can add the setting Kevin mentioned to help troubleshoot, and if your bed is really out of level, that setting will help fix it.