Klipper3d/klipper #4350: Graphs etc in Klipper documentation can barely been seen in dark mode

mathiasm74, 2021-06-05:
The documentation for Klipper has images of black graphs etc what has transparent background, which means they are barely visible in browsers with "modern" dark mode enabled. See example:

[image]

aaaidan, 2021-06-12:
A low opacity white background would provide the necessary contrast in dark mode, without changing the appearance in light mode. 

I'd be happy to edit the pngs this if there's agreement it's a good way forward. 

I suspect these images have source files though. Maybe that's a better place to make the change...

KevinOConnor, 2021-06-12:
Most of the images were created with Inkscape and the source files are in the docs/img/ directory.

-Kevin

Arksine, 2021-09-28:
Ah, IIRC I used draw.io to generate those svgs.  I'll see if I can regenerate them with opaque backgrounds.

KevinOConnor, 2021-09-29:
Thanks.  This issue should be resolved now with the merge of #4736.

-Kevin