Klipper3d/klipper #4351: z_tilt with Z Motors in series

Luke-Elsdon, 2021-06-05:
I'm trying to set up the z_tilt configuration. I am getting the error "z_tilt z_positions needs exactly 1 items"

I'm guessing this may be due me running two z motors in series using an SKR PRO V1.1 board. I only have one z_stepper configured as this controls both motors, I had attempted to clone the settings into a stepper_z1 config but this would error out as the pins were duplicated.

Is there another way I need to set this up?
[klippy.log](https://github.com/KevinOConnor/klipper/files/6603313/klippy.log)

Luke-Elsdon, 2021-06-06:
Closing. Realising where I've gone wrong here..

leandrotoledo, 2021-08-28:
@Luke-Elsdon what was it? I'm getting the same error.

Thanks.

Luke-Elsdon, 2021-08-28:
Apologies for not explaining originally. I had been working myself stupid on my printer and was a bit embarrased by the fix 😛 

In my case, the SKR Pro V1.1 has the ability to drive a dual Z printer on one driver via jumpers. So while I was defining two Z motors, the board was only aware of "one"

The first setup wouldn't have been able to drive the motors individually anyway so while the error didn't tell me exactly what was wrong, it did make sense once I'd taken a break.

I had to rewire the motors to use a driver each and update my Z configs.

leandrotoledo, 2021-08-28:
@Luke-Elsdon ha, I was doing the same thing - nothing to be embarrassed about 😄 Thanks for clarifying! 

I wish the warning message was a little more descriptive though. Maybe you can reopen this bug and the devs can take a look at?

Thanks!

Luke-Elsdon, 2021-08-28:
I'm not sure if that message is made to catch anything else, and I just got it due to an unusual (incorrect) config. If the only way to get that error is by doing what we did then I agree it could be more descriptive.

Reopening for developer's thoughts on this?

closed 2021-08-30 by github-actions[bot]