Klipper3d/klipper #4358: Disabling pressure advance breaks quad gantry level

vstruk, 2021-06-08:
Quad gantry level works very reliably until I disable pressure advance(either via config or issuing a command "SET_PRESSURE_ADVANCE ADVANCE=0.0 SMOOTH_TIME=0")

With pressure advance disabled QGL constantly fails with "Probe triggered prior to movement".

See the attached log file. Steps to reproduce:

1. G32(A macro, essentially the following commands)
    G28       
    QUAD_GANTRY_LEVEL
    G28
QGL finishes successfully.

2. Run "SET_PRESSURE_ADVANCE ADVANCE=0 SMOOTH_TIME=0"
3. G32 fails with "Probe triggered prior to movement"
 
Repeatability: 100%.
Printer: Voron 2.4
Version: be4fb7f128 

A quick debugging showed that the error occurs in klippy/extras/homing.py, line 97, "check_no_movement".

[klippy.log](https://github.com/KevinOConnor/klipper/files/6613423/klippy.log)

KevinOConnor, 2021-06-08:
Okay, but someone would need to troubleshoot why that is occurring.  We don't use github to troubleshoot problems on individual printers.  You can ask on Klipper Discord or Discourse to see if others are experiencing the problem or to see if others have advice.

-Kevin

vstruk, 2021-06-09:
I don't think this is a printer specific thing. It has nothing to do with the particular hardware.
Other than that the printer is fully functional(300+ hours of printing).
It might be specific to the config file parameters though. A corner case, an unlucky combination of speeds and timings.
I've asked another Voron user to test it and he couldn't reproduce it. But his config is apparently differs from mine.
It seems to me as a software bug.
Anyway, I'd be happy to help with debugging. The bug does not affect me since I keep pressure advance enabled. I've just found the bug and created the issue to make klipper better.