Klipper3d/klipper #4367: Small improvement suggested to BLTouch.md

ReXT3D, 2021-06-08:
I suggest the following small clarification be added to the BLTouch.md document:

"_You need to configure these pins according to your wiring. Most BL-Touch devices require a pullup on the sensor pin.  **If the board does not supply an external pull-up resistor on the input used for BL-Touch,**  prefix the pin name with "^" **to use the internal pull-up**. For example:_"

See here for a bit more background: [Duet 3 Mini 5+ 5V supply capacitive load limits](https://forum.duet3d.com/topic/23593/duet-3-mini-5-5v-supply-capacitive-load-limits/5)

Thank you.

KevinOConnor, 2021-06-12:
Interesting.  I'm not sure we want to reword the documentation, because a lot of users wont know if they have an external pull-up and the inverse problem tends to cause many more issues for users.  (Specifically, they leave the line floating and they get random failures from that.)

I am curious to see if your issue goes away with the internal pullup removed.

-Kevin

ReXT3D, 2021-06-12:
My suggestion was purely based on the fact that I suspect many people simply don't know what they don't know.  In my case for example, when I put together my printer.cfg I read the full content of BLTouch.md and I immediately assumed that I need the pull-up as referenced in the document.  Now, I am not a 3D printing or MCU expert, but I am very comfortable with all things mechanical and electrical (engineer with 25+ years of work).  I thought that highlighting the fact that the pull-up is required but not on all boards would at least make people think about it and perhaps ask questions.  But I do understand your point.  In my case the outcome was the same - three head crashes into the bed that were arrested by the Z steppers stalling on torque.  The first one was not with reduced current, unfortunately, and I had to replace the X carriage wheels as they developed permanent indentations (I have some pretty screenshots of the resultant bed mesh waviness).

It has been a while and I have done a huge amount of experimenting trying to trigger the misbehaviour with no success.  So it is looking very positive.  I was fairly certain that the problem was fixed once I looked at the electrical noise on the BL Touch lines with my scope, but it's nice to have the empirical proof.

In the end I did four separate things:

1. Routed the heater and stepper wiring in one bundle (high current) and the BL Touch, fans, thermistor and stop switches in a second bundle, both spaced at least 2 inches apart.
2. Added an internal ground reference wire between the main 24V printer supply and the Raspberry Pi 5V supply.  The original setup worked flawlessly with the Creality Atmega board for almost a year, but I suspect that everything became more noisy with the high performance Duet and referencing the grounds through the USB cables just seems wrong.  Even though everyone who is using an external Raspberry Pi power supply does exactly that.
3. I disabled the internal pull-up resistor as discussed.
4. I added a reasonably low impedance 33 uF capacitor near the BL Touch.  Even though dc42 strongly suggested that it is not required, I feel that it will provide a good reservoir for the initial inrush current associated with BL Touch pin movements.  Note that my MCU to BL Touch wiring is longer that 1.5 meters and uses mostly 26 AWG conductors.

On a separate note, the Duet 3 Mini 5+ is awesome!

Peter.

ReXT3D, 2021-06-25:
Quick update: after many prints and countless hours of me trying to purposefully trigger BL Touch misbehaviour, I can confirm that everything has been working perfectly.

Sineos, 2021-06-25:
I guess I can somewhat confirm @ReXT3D findings.

I'm using a BTT SKR1.4. When connecting the BL-Touch to the dedicated Probe-Pin on the board, the BLTouch will not work until the pin is pulled-up (^) in the cfg (took me a while to figure this out). It though works out-of-the-box on a regular Endstop pin

See the relevant schematics:

[image]

Talha909, 2021-08-05:
> I guess I can somewhat confirm @ReXT3D findings.
>
> I'm using a BTT SKR1.4. When connecting the BL-Touch to the dedicated Probe-Pin on the board, the BLTouch will not work until the pin is pulled-up [...]

Can you please upload the config file. My Bltouch is not working on probe pin. I did all the things

Sineos, 2021-08-05:
```
[bltouch]
sensor_pin: ^P0.10
control_pin: P2.0
pin_move_time: 0.680
stow_on_each_sample: True
probe_with_touch_mode: False
```

BLTouch v3.1

closed 2021-09-17 by github-actions[bot]