Klipper3d/klipper #4373: position_endstop in section 'stepper_z' must be between position_min and position_max

EzrielDreamurr, 2021-06-09:
Having issue configuring klipper

when i typed a command in octoprint, i recieve the followings
Recv: // position_endstop in section 'stepper_z' must be between position_min and position_max
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: !! position_endstop in section 'stepper_z' must be between position_min and position_max
 
The config is here:

[stepper_z]
step_pin: PB7
dir_pin: !PB6
enable_pin: !PB8
microsteps: 16
rotation_distance: 8
endstop_pin: probe: z_virtual_endstop
position_endstop: 0
position_min: -2
position_max: 250

and i heard that i need to upload klippy log, and i don't know how to.

zellneralex, 2021-06-09:
The error message is wrong. As you use a virtual_endstop you are not allowed to specify an position_endstop.

Yoyolick, 2021-06-10:
> The error message is wrong. As you use a virtual_endstop you are not allowed to specify an position_endstop.

Im very confused by this same issue right now. Going off of what you said, when I remove `position_endstop: 0` from my cfg I still get `position_endstop in section 'stepper_z' must be between position_min and position_max`, and if I remove the `position_max` from my z stepper i get  `position_max' in section 'stepper_z' must be specified`

Yoyolick, 2021-06-10:
I think i figured it out, this seems to be an error applied by multiple different things, for me it was having a negative bltouch Z offset. So make sure you have a positive Z offset if you are seeing this and have this problem

closed 2021-06-11 by github-actions[bot]