Klipper3d/klipper #4375: "emergency stop" terminology is misleading

5PAZT1C, 2021-06-10:
please see a similar issue raised with marlin firmware here:
https://github.com/MarlinFirmware/Marlin/pull/15065

By definition Estop needs to kill all motion power. Machine must be in a locked state.

Since the term emergency stop may also implicate electrical requirements depending on where its used, should we relabel this? I have yet to see a printer board compliant to PLd / Class 3 safety....

Note : what we have here is at BEST case PLb with a category 2 stop. With power cutoff to steppers and heaters we get closer to a category 0 stop but still PLb. Emergency stop needs to be category 0 or 1 and typically is PLd or e

ISO 13850 limits the selection of stop category to Category 0 or 1 and excludes Category 2. This exclusion can be found in NFPA 79, IEC 60204 – 1, and CSA C22.2 No. 301 as well. Category 2 may only be used for operational or “normal” stopping functions.

https://machinerysafety101.com/2010/09/27/emergency-stop-categories/?doing_wp_cron=1566846639.6410090923309326171875

One more critical paragraph to note :
Both ISO 13849 – 1 and IEC 62061 [8] base the initial requirements for reliability on the outcome of the risk assessment (PLr or SILr). If the stopping condition is part of normal operation, then simple circuit requirements (i.e. PLa, Category 1) are all that may be required. If the stopping condition is intended to be an Emergency Stop, then additional analysis is needed to determine exactly what may be required.

Note that PLa Category 1 is NOT considered an Emergency stop.

5PAZT1C, 2021-06-10:
One possible solution could be to rename the currently implemented system to "immediate halt" instead of "emergency stop"

KevinOConnor, 2021-06-12:
Okay.  We don't use the term "emergency stop" that frequently in Klipper.  It's fine if someone wishes to update the documentation and submit a PR.  (We don't use github for feature requests, though, so if you're looking for someone else to make changes you can try asking on Klipper Discord or Discourse.)

-Kevin

5PAZT1C, 2021-06-12:
thanks, perhaps this is more relevant to fluid and mainsail?

closed 2021-06-12 by 5PAZT1C