Klipper3d/klipper #4380: z_offset gets ignored above a certain value

chunter1, 2021-06-12:
I got the issue that when trying to lowder the nozzle by increasing the z_offset value, all values above a certain value are getting ignored.
For example....

- My standard BLTouch z_offset is 1.2 mm
- When i try to lower the nozzle by incresing the value to 1.3 mm it still prints with exactly the same first layer height!
- Whereas when i set the z_offset to e.g. 1.0 or even 0.0 mm it raises the nozzle accordingly above the bed.

Why are z_offset values greater than (approx.) 1.2 mm ignored wheras lower ones accepted?

Please find attached my config.
[klipper_config.txt](https://github.com/KevinOConnor/klipper/files/6641894/klipper_config.txt)

phil2sat, 2021-06-12:
Don't know but you're missing "#position_endstop:" Value.
mine was defaultet to position_endstop: 0.5 and z_offset ist working fine.

chunter1, 2021-06-12:
> Don't know but you're missing "#position_endstop:" Value.
> mine was defaultet to position_endstop: 0.5 and z_offset ist working fine.

When i enter a value there i get an error message saying that inconjunction with a BLTouch probe that value must not be set.

KevinOConnor, 2021-06-12:
Sorry, but we don't use github to help troubleshoot issues.  If you'd like to track down why this is occurring on your setup then feel free to update this ticket, otherwise if you're looking for help you should follow up on Klipper Discord or Discourse.

-Kevin

closed 2021-06-14 by github-actions[bot]