Klipper3d/klipper #4384: Missing info in description for bed_mesh Move Splitting

ld3300, 2021-06-13:
On line 172 of [klipper/docs/Bed_Mesh.md](https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Mesh.md#move-splitting) under the description for "### Move Splitting"

Current sentence: "Long moves must be and split into smaller moves
to correctly follow the shape of the bed."

It feels there is a word or statement missing in this location: "Long moves must be (????????) and split into smaller moves
to correctly follow the shape of the bed."

-David

[klippy.log](https://github.com/KevinOConnor/klipper/files/6643128/klippy.log)

Sineos, 2021-06-14:
I'd guess rather the "and" is surplus

KevinOConnor, 2021-09-27:
I reworded that paragraph (commit 6e87befa).

-Kevin

closed 2021-10-05 by github-actions[bot]