Klipper3d/klipper #4393: DS18B20 unable to read crashing Klipper

jvinraviv, 2021-06-14:
I've had this happen a a couple of times now where for some reason Klipper can't find the DS18B20 sensor and resets the FW.
Problem is when this happens in a print it is ruined.

In the attached file the crash occurred on line 1659 while I was running a PID tune.

Is there a way to prevent a misread of the sensor from resetting the FW?

[klippy.log](https://github.com/KevinOConnor/klipper/files/6650199/klippy.log)

closed 2021-06-16 by github-actions[bot]

screech24, 2021-11-03:
I having the same issue and I've done everything possible to make sure that it's not a hardware related issue. Would be nice if Kevin didn't just disregard small issues like this and looked into them further:( I've asked in the discord server several times and have had no response to this issue.

charredchar, 2022-02-02:
I am having the same exact issue. I've had it happen once during a print but multiple times outside of printing. I have moved away from trying to use the DS18B20 for anything more than monitoring temps around my printer and not tie it to any device as it doesn't seem reliable enough within Klipper.

There is another issue open related to it failing for a heater, it seems Klipper is just not happy with this sensor. I think the next best thing would be to have an option within Klipper that, when used as just a `temperature_sensor` and not tied to a device then set it to not cause a firmware error. If this is not a viable option then the use of this sensor should probably just be removed from Klipper until it can be reliable. The fact it is mentioned in the config reference led me to believe I could use it as I would any other sensor and purchased a handful, now none is usable in Klipper.