Klipper3d/klipper #4397: [Bug] Tmc2209 in uart mode (register errors)

Ramalama2, 2021-06-14:
Hi, just wanted to open here an issue for this, because many people already complains in discord about this.

To make it short:
Everyone that is affected uses so far tmc2209 in uart mode.

The issue basically for most of us:
If the motors are powered off, for example after a printer start, and you try to home or move the motors, you get all possible register errors (can't read/write trsync_state/ifcfr/and many more)

For some of us it gets fixed by adding:
```
[gcode_macro TURN_ON_MOTORS]
gcode:
  SET_STEPPER_ENABLE STEPPER=stepper_y ENABLE=1
  SET_STEPPER_ENABLE STEPPER=stepper_x ENABLE=1
  SET_STEPPER_ENABLE STEPPER=stepper_z ENABLE=1
  SET_STEPPER_ENABLE STEPPER=extruder ENABLE=1

[delayed_gcode AUTOSTART]
initial_duration: 1
gcode:
  TURN_ON_MOTORS
```

For others, it doesnt help sadly.

For me at least that gcode fixes the issue.
(I can't say if it's after the update, or was before there, because i personally switched to the btt e3 rrf after the update. But others complains about this, after the last firmware update)

I will attach my log, after i finished the print.

Thanks, if i missed something, please tell.

techdaddy1990, 2021-06-15:
i had the same issue with using 2 SKR2s i will attach my log for the cause.
[klippy (7).log](https://github.com/KevinOConnor/klipper/files/6656445/klippy.7.log)

techdaddy1990, 2021-06-15:
unfortunately the [gcode_macro TURN_ON_MOTORS] did not help me.

matthewlloyd, 2021-06-15:
Make sure you have read the discussion here in case it's relevant: https://github.com/KevinOConnor/klipper/pull/4234?fbclid=IwAR2ivKQvbu1wVicf4gG56bEDjiBOvUT8SKAok8-cDIya2YrQAarny8BX--c

Ramalama2, 2021-06-15:
> Make sure you have read the discussion here in case it's relevant: #4234

That's a good hint, many in discord used an skr2 indeed.

But i am using here an btt e3 rrf v1.1 that doesn't has the skr2 first Revision bug.
Like i mention, for me for example, without the macro above, I don't get klipper working at all, basically you turn on the printer, and errors...

But i tryed marlin & reprap, because this issues drived me that crazy, that i thinked I can't use klipper anymore, until i found that out.
However, marlin & reprap, doesn't have any problems and i don't need there to explicitly enable the drivers or anything, like i do with the macro above.

What wonders me most is, that people complain in the discord and i tagged them to report the issue here and opened this whole issue thing, now no one writes 🖕👍

However, my log is comming, just sorry, i have a 2 day print running, but it's done in some hours 🙈

Cheers and Thanks 👍

KevinOConnor, 2021-06-16:
Sorry, but we don't use github to help diagnose problems with a printer.  Feel free to raise on Klipper Discord or Discourse.

-Kevin

Ramalama2, 2021-06-16:
[klippy_1.log](https://github.com/KevinOConnor/klipper/files/6662358/klippy_1.log)
--> This log is where it immediately crashes on power on and G28

[klippy_2.log](https://github.com/KevinOConnor/klipper/files/6662360/klippy_2.log)
--> Here the G28 worked once, but after turning off the motors and G28, the issue happens again.

Here are the logs from me.

> Sorry, but we don't use github to help diagnose problems with a printer. Feel free to raise on Klipper Discord or Discourse.
> 
> -Kevin
Diagnose Problems? Keven, this is definitively a klipper bug, since marlin and reprap doesn't have this issues.
I don't know if klipper powers the motors before moving them or not to activate the drivers, but on the other board with standalone drivers, this problem doesn't appear.

lunatic1972, 2021-06-16:
I am having the exact same issue and also disagree Kevin, you must first be able to power on and actually get to print to diagnose a 3d printing issue. This is not the case since it never gets that far and fails not long after boot. I'm currently using mainsail with latest klipper as of June 14th. I only started having the issue after updating klipper to latest version(June 14th). I could go back but to me, it seems like something has changed. If I need to modify the printer.cfg or TMC2209 drivers that's fine but please at least look into it and if it's a user change that can address it maybe add it to the list of changes. TIA

ReXT3D, 2021-06-16:
For what it's worth, Duet 3 Mini 5+ with 2209s in UART mode is working perfectly here...

lunatic1972, 2021-06-17:
I went back to May 4th commit and now everything is back up and running. I didn't try any in-between more so because of time constraints but it may give someone some insight as to what may have changed in that time frame

closed 2021-06-19 by github-actions[bot]

DaviVolcan, 2021-06-26:
i was having this problem too... it stoped with Ramalama2 script... but now my drivers are always on... i m suposed only to solve bugs at github? i shouldn't report them?

AndrezinhoCosta, 2021-06-26:
Unfortunately I'm having the same problem! I'm using SKR Pro 1.2 with TMC2226 in a RatRig V-Core 3 with Fluidd.
I've trying to solve with several modes, but nothing is working.
Just now the  [gcode_macro TURN_ON_MOTORS] doesn't work...
Maybe have I to use Marlin again to solve my problem? I don't wanna do this!

screech24, 2021-06-26:
I am Also having the same issues with an mks genl v1.1 avr based board with tmc2226 in uart mode. So it's not board specific but uart mode issue. If this doesn't get acknowledged as a bug soon I'm sure quite a few of us will be going back to marlin. Getting sick of errors and not even being able to use my printer.

jim-sokoloff, 2021-06-28:
I may have a significant discovery here. I ran into this error tonight as part of commissioning my newest printer. It's a core-XY using TMC2225s (2208's) in uart mode on a BTT Octopus v1.0.
I ran into this error pretty repeatably while trying to get Z_TILT_ADJUST dialed in. It would die on the 2nd pattern typically.
I found this thread, applied the TURN_ON_MOTORS gcode from @Ramalama2, no help.
I then realized that the X&Y (A&B) stepper motor driver heatsinks were too hot to touch. (I'd moved the fan aside while doing related troubleshooting to get the board brought up.)

### Possible solution found:
I put a fan back on the heatsinks and could run dozens of Z_TILT_ADJUST cycles.
I then took out the TURN_ON_MOTORS, repeated the tests, and everything worked.
I then did some unrelated printer work and klipper configuring.
I then came back, removed the fans, and could induce the failure again after about 3-4 cycles. (I suspect there's some heating above ambient that the drivers can take just passively cooled, but once you get them hot, they fail sooner on the next run.)

Anyway, if you're seeing this error, improve the cooling on your stepper drivers and see if that eliminates it.
I'm ~97% certain that my issue is tightly related to heat. (This could also explain why some people run into it and others haven't seen it.)

Ramalama2, 2021-07-31:
@KevinOConnor Since some days, this error doesn't happens anymore. I just gaved it another try without my Turn_on motors macro.
It's definitively resolved now.
Idk which commit fixed it now, but just wanted to give a feedback.
Cheers ✌️

KevinOConnor, 2021-08-01:
Thanks.  It's likely #4505 that fixed it.  @ReXT3D did the work to track down an issue with tmc2209s and shared uarts.

-Kevin