Klipper3d/klipper #4401: Klipper Documentation - Improvement greatly needed

jatmn, 2021-06-15:
I understand Klipper is not meant for the average user and meant for the power users that understand the core components of machines at the hardware and software levels that are seeking to gain as much potential as possible out of their hardware.

However, both Marlin and RepRapFirmware have set the bar quite high when it comes to documentation (in recent years) which helps when their power users and machine modders come looking for it. 

There is a considerable push by users of Klipper firmware that it is the Best and there is no reason to use other firmware options and with the increased popularity of the Voron 3dprinter design Klipper is getting a lot more attention.
This has caused a very.. let's say unfavorable image of the Klipper community in some ways. 
A large part of this is due to the complexity and confusion the documentation causes, and when people ask or voice concern about it on public channels they usually met with a Cult-like response of bulling.
I will say, however, your unpaid team and users on the Discord Server are VERY understanding and friendly which was quite refreshing to find. 

After a rather lengthy conversation with several **Helpers** and **Mods** on the Klipper Discord, it was recommended I open this ticket.

The Klipper Documentation from Overview to Installation Guide to Configuration Resource documents could really use some sprucing up in the following ways. This is applicable to all Guides, Configurations and associated Resources and this list of examples is by no means conclusive.

### There is a lack of consideration of users' level of hardware knowledge (intentionally?) and makes broad assumptions

-  _Example:_ **Overview**
 Completely lacks a formal list (or link to) of supported controller boards. It assumes the user knows to search for firmware support purely by MCU or to dig thru the example configs. Though in this it also does not actually direct the user to check example configs either just vaguely lists it at the end

### There is for the most part a complete lack of inline cross-references to basically everything.
-  _Example:_ **Installation**:
Completely lacks a formal link to supported controllers which I feel should be reiterated in this section as many would just jump to this section ignoring Overview. The guide assumes the user already knows their board is supported and committed to Installing without a link to the list of controllers or a screenshot to show the user what they are going to be expected to deal with. Which does not allow the user to consider if they are up for this undertaking or not. Overall the guide feels it was put together in a very rushed matter, it could use some work.
   
- _Example:_ **Config Reference - [stepper]**:
This section **_DOES_** go into considerable detail explaining the various parameters, and at the start of the section even links to "rotation distance document" to help the user with calculations. However, this section completely lacks any notes or preferably a link to the required sub-section for users using TMC drivers. The user is left to discovering this later if they read the whole document and scan the section's list.
 
- _Example:_ **Config Reference - TMC stepper driver configuration**
This section might mostly be due to order and formating.. but its easy to skip over the link to "TMC Drivers guide" if your looking for details for a specific driver series first. I actually completely missed this link several times because I just jumped to the 2208 and 2209 drivers then got real confused at the lack of reference on how to configure sensorless homing and had to scroll all the way to the start of the section to find it.  Also side not on this section why is tmc2130 before  2208.. when the rest of the section appears to be in numerical order? (consistency issue)

### Recap
Overall the documentation heavily assumes the user has already read every possible bit of documentation before getting started _(many users do things in steps/sections at a time so this is very bad)_ It also makes broad assumptions that the user is just going to Ctrl-F (search) for keywords on the document to track down the information they are looking for anytime something is unclear or they need additional details rather than being directed to it.

This leaves even people very experenced with configuring Marlin , RepRapFirmware, or any other firmware.. Quite frustrated when looking for information. 

A common response I have gotten was to submit PR's to fix the issues I see in the documentation, however... If I don't even understand what I am reading or where to find the associated reference information that's lacking, how am I qualified to make PR for it? This needs to be taken on by someone or group of people that know their way around the firmware quite well.

Something needs to be done because the noisy few visible from the outside of the Klipper community give the image that the firmware is for the elitist and if you ask questions out of confusion you get trampled, and as the documentation CAUSES questions many people avoided the firmware completely never even giving it a chance.

AndrewEllis93, 2021-06-16:
> they usually met with a Cult-like response of bulling.

> This has caused a very.. let's say unfavorable image of the Klipper community in some ways.

> noisy few visible from the outside of the Klipper community give the image that the firmware is for the elitist and if you ask questions out of confusion you get trampled

I really don't see how any of that is relevant to a GitHub issue.

jatmn, 2021-06-16:
> I really don't see how any of that is relevant to a GitHub issue.

yes because that was clearly the most important part of all this *facepalm*

AndrewEllis93, 2021-06-16:
No, you bring some valid points, it could just do without the weird interjections of drama and hyperbole. I don't see how any of that is constructive or relevant. 

The documentation could use some love, yes. I think ultimately it's just not quite as mature as some other firmwares. One thing I would absolutely like to see is some sort of hierarchy to the documentation, rather than a monolothic folder of readmes.

KevinOConnor, 2021-06-16:
Okay.  However, this looks like a "feature request" and we don't use github for feature requests.  If you or someone else is actively working on a revised documentation then feel free to post the status of that work here on github.  Otherwise, feel free to raise on Klipper Discord or Discourse.

-Kevin

jatmn, 2021-06-16:
> Okay. However, this looks like a "feature request" and we don't use github for feature requests. If you or someone else is actively working on a revised documentation then feel free to post the [...]

Your staff discord literally told me to post it here. lol

Gilabite, 2021-06-16:
"staff" lol

AndrewEllis93, 2021-06-16:
> Your discord staff literally told me to post it here. lol

Staff? It's an open source project. As Kevin mentioned, you can contribute to it.

jatmn, 2021-06-16:
> > Your discord staff literally told me to post it here. lol
> 
> Staff? It's an open source project. As Kevin mentioned, you can contribute to it.

The people on discord the roles "mods" and "helpers" assigned to them.

And if I knew enough about the firmware to make the contributions myself I may, however I don't which is one of the reasons why I brought it up.

jatmn, 2021-06-16:
I can see my feedback is not welcomed here, this is closed.

closed 2021-06-16 by jatmn