Klipper3d/klipper #4407: Klipper - Unable to obtain 'trsync_state' response [bug]

Murilolana, 2021-06-18:
Hi,

After updating klipper to the latest version, my printer stopped working, reporting the "Unable to obtain 'trsync_state' response" whenever I tried to run a gcode or a macro. It didn't happen if I issued a G28 or a single axis movement, but running a gcode would trigger this error and shutdown the mcu. I tried disabling almost all octoprint plugins to check if any of them were overloading the mcu, but that made no difference. I've searched over the discord server for people having this problem and found a few that reported this after updating to the latest version some days ago. So I downgraded to the commit 099a6dd, reflashed the firmware and my printer works flawlessly again.
I'm using a SKR Pro v1.1 board with TMC2130 drivers on SPI. Apparently, the other users experiencing this issue a running TMC2209 on UART.
I lost the klipper.log file after downgrading. I would have to reinstall the current version to reproduce the problem.
Regards,

Murilo Torres
[log.zip](https://github.com/KevinOConnor/klipper/files/6678881/log.zip)

Murilolana, 2021-06-19:
I didn't request help, just opened this issue to alert that the last commit caused the reported problem. After spending the whole day diagnosing this, I thought it would be helpful for others going through the same issue and of course, the developers. I don't think the "invalid" label is appropriate. Apparently, the last commit messed with SPI and UART drivers.
Regards.

Ramalama2, 2021-06-19:
https://github.com/KevinOConnor/klipper/issues/4397

Here is the same, read kevins answer 👍
While on one side he did awesome with making klipper and bring it to life and all the work etc... And i probably can understand, that this is overwhelming and nervy to do this all almost alone, I can't agree how he handles issues here. Or with the way of disrespect.

However, you can try the enable motors gcode from the other issues thread and if that doesn't help, you can revert the one commit, while still staying up2date:
git pull (to update to latest)
git revert commit (to revert that one commit)

I think the breaking commit is this one:
https://github.com/KevinOConnor/klipper/commit/e1f7748e1a9dfda3e78f8a27e9972a9eb954e946
That would mean:
git revert e1f7748e1a9dfda3e78f8a27e9972a9eb954e946

Cheers

oscaroecm, 2021-06-19:
With all due respect, I believe you are completely misunderstanding what is the gihub/klipper for and even worse, feeling disrespected by the answers you are receiving. As far as I understand, you are being told that problems with printers, bugs, requests for help etc. are to be posted elsewhere. Post your issues where apropiatte and you are likely to get a response. You are just being told that here you won't. Nothing rude or inapropiatte there.

#4397

e1f7748

Cheers

lpomykal, 2021-06-19:
Hi guys,
I have the same problem.

It occurs while using my DeltaCalibration macro:

[gcode_macro DeltaCalibration]
gcode:  
        M104 S200 ; set extruder temp
        M140 S60 ; set bed temp     
        G28   
        M109 S200 ; wait for extruder tempG28
        M190 S60 ; wait for bed temp
        #M300
        DELTA_CALIBRATE
        #M300
        SAVE_CONFIG
        #M300 
        G28         
        #M300       
        #M300
        
If I use only manual commands for temperatures + MacroHome and send a DELTA_CALIBRATE command, it works.

[gcode_macro MacroHome]
gcode:  G28
        #M300

[klippy.log](https://github.com/KevinOConnor/klipper/files/6680650/klippy.log)

Ramalama2, 2021-06-19:
Bugs belong into github issues, there is the exact problem and logs attached and more as one reports this issue, since the firmware update.

So what is wrong here, what is missing? The exact line which causes the bug?
https://www.klipper3d.org/Contact.html
Mentions about a klipper "community forum", which doesn't exists and is basically a discord room full of help requests and no one to actually help there?

And by the way, this issue was discussed in the klipper discord, aka klipper "forum", with no results at all, that was the reason, I've opened the first issues thread in github.
Just read the first post from the original issue, that was closed.
But i guess, since you guys don't have that bug, you don't care and don't want to hear it 👍

It's useless anyway, enjoy your weekend 👍

Murilolana, 2021-06-20:
This is the exact place to report this kind problem, like pointed by Ramalama2. And I disagree that the place for help request should be elsewhere either. At least it should not be the discord server, where your message gets lost over tons of newer ones. That's a chat platform, not a forum. The github structure is perfect for this, where there are all the tags like "open", "closed", "solved" and so on. Your issue doesn't get lost here and is easily found on Google search.

Klipper is an awesome software and I'm sure it will keep being, as long as bug reports are not misunderstood as plain help requests.

Regards.

BlackStump, 2021-06-20:
I do not see that the instructions to report a issue have been followed - [Please re-read and provide required info](https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#i-have-diagnosed-a-defect-in-the-klipper-software)

Where is your proof that it is a actual bug and still not a user issue?

Do not use Marlin as a example - Klipper is not a Marlin Clone.

Sineos, 2021-06-21:
How Github is used is finally the decision of the project owner. 

As detailed here https://www.klipper3d.org/Contact.html, the preferred way is to open a topic in the Discourse Server (https://klipper.discourse.group/) or seek help in Discord (https://community.klipper3d.org/) until there is an agreement that it is really a bug.

Especially the Discord server is read and monitored by Kevin, so if you really seek for help, then please follow the described way.

Murilolana, 2021-06-21:
I think we're all wasting our time here. We're disussing wheter the format of a proper bug report was fully filled or not. This is useless. I've read the directions on steps to report this. I've searched over the discord server before opening this issue and all I've found were few lost messages under random discussion. I made sure that "there is a problem, I can't figure out why, and therefore it is a Klipper bug" pointed was not my case. Spent the whole day investigating this, as mentioned before.

I've updated the firmware and it stopped working, downgraded and it worked again. Other user reported the exact same thing. Plain and simple, the difference is in the code, not the hardware.

> Where is your proof that it is a actual bug and still not a user issue?

Is this really a matter of proving I'm right and you are wrong? We are on the same side here! Jesus!!!
I love using klipper and really want it to grow much beyond marlin. I've opened this issue to contribute for that, not to trash the firmware. In fact, I intend to become a donator soon enough cause I don't have any plans on turning to a different solution and I know there are people spending time on klipper to make it even better.

I'll discuss this no more. Let's move forward.
Regards.

Ramalama2, 2021-06-21:
Yeah, you realized it, im out here too, this was just stupid to open here a bug report.
Lession lerned, waiting for rrf with input shaper and bye.

Sineos, 2021-06-21:
> Is this really a matter of proving I'm right and you are wrong?

No it is not. It is simply a matter of respecting the decisions of the project owner who is developing and providing this at no cost and at no obligation for anything.

To your special problem:
Check the discourse server. You will notice quite a few TMC error posts. 

Around the 20th of May and following, Kevin improved the error detection of the TMC drivers and made Klipper to error out instead of silently ignoring errors.

This lead to a significant increase of reports. Many of them reported success by changing the TMCs, thus getting rid of the problem.

So finally we come back to the question: Is it a Klipper bug or is it still the possibility that your setup suffers from a defect driver, unstable power-supply, fishy cabling or what not? An hardware error that simply gone unnoticed prior to the changes?

Murilolana, 2021-06-21:
> > Is this really a matter of proving I'm right and you are wrong?
> 
> No it is not. It is simply a matter of respecting the decisions of the project owner who is developing and providing this at no cost and at no obligation for anything.

I was being sarcastic, got it? I do respect Kevin decison, but I don't agree it's the best way to deal with it. Read the thread again.

> To your special problem:
> Check the discourse server. You will notice quite a few TMC error posts. 
Read my last comment again, that's what I did. By the way, if a group of people start reporting the same problem after a firmware upgrade, it's not one's special problem... 

> So finally we come back to the question: Is it a Klipper bug or is it still the possibility that your setup suffers from a defect driver, unstable power-supply, fishy cabling or what not? An hardware error that simply gone unnoticed prior to the changes?

No defect driver, expensive certified PSU, no shitty cables, 4 identical printers with the same setup. Again, (for the last time) spent the whole day testing and isolating the cause. Checked all the cables, all connections, different set of motors, temperature sensors, different RPI4, fresh new octoprint image on different sdcard, reflashed the bootloader with stlink, and so on... I really investigated this before opening this issue. Maybe you got offended cause I added "[bug]" in the title, but again, read the thread and you see that my intention is not to trash the firmware (I've already said that).

TheFanatr, 2021-06-22:
If I may add something, I think it would be in everyone's best interest if future conflicts of this nature are potentially largely avoided via not labelling issues of this nature with `invalid` but rather something like `unknown-cause` or `possible-user-error`, or something to that effect, giving people the green light to find out whether or not it is user error (setup-specific misconfugration, such as a faulty or poor driver, faulty or poor wiring, something disconnected, something done wrong) or if it genuinely is the result of a bug. The reason this would be useful is because if a significant number of people are doing the same thing incorrectly it might be reasonable to resolve the issue with additional documentation, or a note, or something of the like. It will also decrease the apparent hostility between select community members (and maybe better support from the community).

Ramalama2, 2021-06-22:
> No it is not. It is simply a matter of respecting the decisions of the project owner who is developing and providing this at no cost and at no obligation for anything.
>
> To your special problem: [...]

Does powering the drivers with my workaround gcode, before using them, sounds like a wiring or a driver problem?
I mean this:
```
[gcode_macro TURN_ON_MOTORS]
gcode:
  SET_STEPPER_ENABLE STEPPER=stepper_y ENABLE=1
  SET_STEPPER_ENABLE STEPPER=stepper_x ENABLE=1
  SET_STEPPER_ENABLE STEPPER=stepper_z ENABLE=1
  SET_STEPPER_ENABLE STEPPER=extruder ENABLE=1

[delayed_gcode AUTOSTART]
initial_duration: 1
gcode:
  TURN_ON_MOTORS
```
That snippet of gcode, solves all my issues luckily, no errors, nothing, 24h prints does perfectly fine. Without it, my printer is completely unusable.

I don't want to blame anyone, just in my eyes the handling of bug reports here feels totally wrong.
I mean, i understand completely that there are many "newbies" and people that "would" or are reporting nonsense (user errors).
But this here is clearly not this case with newbie reports, or lets say, this is something that has a high chance to be a bug and doesn't looks like an user error.
But no one wants to even hear of it and teeling us basically to f..ck off.

And please don't tell me, going to discord is not a way of saying f..ck off. I am in that dicord room, since a long time, tryed to talk about this, but there are only "newbies" and 2-3 members that have clue but doesn't answer.
The messages goes down, and you can't discuss anything properly there.
Im still in that discord, if someone wants to talk about it -> Ramalama
Just mention me.

While on the other side, this here is more like a forum, and in github, we can actually discuss this things, even if the replyes takes very long.

So don't take me wrong, im just pissed, that everyone is just telling me to f..ck off from here and yeah, i lost my motivation for solving this anyway, since my gcode snippet above, solves the issue for me anyway 🤷✌️

KevinOConnor, 2021-06-23:
> And please don't tell me, going to discord is not a way of saying f..ck off.

@Ramalama2 - it seems you are frustrated, and you've posted several times here on github to express your frustration.  This is not the place to do that.  We use github to share the results of work done to improve Klipper.  If you want to track down why your issue is occurring then feel free to post your results here.  Otherwise, please refrain from posting here again.

If you're looking for help, then you can post on the Klipper Discourse server or Klipper Discord to see if there are others having the issue and to see if others can help.  Alas, I suspect no one knows the reason why you're having issues, and it may take some time before someone can assist.

-Kevin

closed 2021-06-24 by github-actions[bot]

Tahutipai, 2021-09-01:
Hi @Ramalama2,

Sorry for the frustrations you experienced here.

FWIW I just wanted to thankyou for taking the time to post your solution here.  I had the same problem, tried many fixes to no avail, but yours worked and solved the problem.

Board: Bigtreetech GTR, w/ BTT 5160 TMC's, up to date klipper v0.9.1-759-gdcf8cb82

You saved me many frustrations with your post.

Cheers.

> Does powering the drivers with my workaround gcode, before using them, sounds like a wiring or a driver problem?
> I mean this:
>
> ```
> [gcode_macro TURN_ON_MOTORS]
> gcode:
>   SET_STEPPER_ENABLE [...]

Ramalama2, 2021-09-08:
> Hi @Ramalama2,
>
> Sorry for the frustrations you experienced here.
>
> FWIW I just wanted to thankyou for taking the time to post your solution here. I had the same problem, tried many fixes to no avail, [...]

Hey, this issue is already solved in the new versions. It was indeed a bug. However, you can try now without my workaround above, all should be fine. (For me at least) 🙈