Klipper3d/klipper #4409: ‘i2c_config’ has incomplete type      struct i2c_config i2c_config;

outbackdingo, 2021-06-19:
Compiling out/src/i2ccmds.o
src/i2ccmds.c:14:23: error: field ‘i2c_config’ has incomplete type
     struct i2c_config i2c_config;

TheFanatr, 2021-06-22:
Why did you close this? I have the same problem; did you find a solution or is this effect due to misuse? If it is the latter, could you please share what you did differently to make it work? Was going to open an issue myself.

outbackdingo, 2021-06-22:
make clean && git pull && make menuconfig again... worked for me.

TheFanatr, 2021-06-22:
I just PR'd it so that people know what to do until someone fixes the `menuconfig` script.

TheFanatr, 2021-06-22:
@KevinOConnor I disagree that this is invalid, because the instructions in the installation documentation are incomplete, and as a direct result a lot of beginners have this same issue until they read the comments of #3682 or happen to read the "How to Upgrade" section of the FAQ which has users perform the step described in the aforementioned comments. I should also note that the answer (mentioned in all of the places noted previously), provided by @outbackdingo, is a sufficient workaround but this does not make the issue fixed, or make it not an issue, or instead make a help request; every user who does not perform a `make clean` in between `make menuconfig` and `make` will have the build fail on them, yet this is exactly what the documentation tells them to do in a vary obvious and visible place. I ask that you reconsider this decision.

closed 2021-06-24 by github-actions[bot]

redfox05, 2021-11-26:
Just to note, I also came across this during a fresh install on octopi Raspberry Pi 3, which crashed during the `make` command. Running the clean instructions above allowed things to continue as normal.

trek-e, 2022-01-02:
Still an issue, just ran into this on my RPI 3 as of Jan 1 2022.

MattEvansA45, 2022-01-24:
As above, I've just encountered this issue following the steps on the main installation page.
[image]

TheDopple, 2022-04-30:
FWIW, this worked for me:
[image]

rsilvers129, 2022-05-12:
Had this issue today so it is still not fixed.