Klipper3d/klipper #4410: Documentation-referenced config/sample-pwm-tool.cfg missing

johnsonm, 2021-06-19:
Commit 88f6061cd references `config/sample-pwm-tool-cfg` which has a typo (`-` for `.`), and is also missing. I have found https://github.com/Cirromulus/klipper/blob/pwm_sync_channel/config/sample-pwm-tool.cfg which I presume to be the closest thing but I get `Error: Section 'menu __main __control __toolonoff' is not a valid config section` — I'm very new to Klipper so I don't know whether that's due to not having configured a display yet (not clear that I even want one) or is related to other function still being worked on a branch.

@Cirromulus — is this an easy sample config to make for the parts of your work that have been merged so far?

Thanks!

Cirromulus, 2021-06-21:
Yes, I forgot this file in my pull request :o
>I'm very new to Klipper so I don't know whether that's due to not having configured a display yet 

Yes, this is the source of your error. In this example config-file, that part added a section to the menu to manually adjust the laser power.
If you copy the file without the two sections, it should work.

As stated in the WIP-PR, you should use your own Pin number and handle the Laser as if it was always on as soon as the printer gets power (safety goggles!).

Also, due to bugs in my WIP, it will help disabling your heaters and other parts of the configuration file that are not needed for the laser. I suggest using two configuration files and starting your printer with either of them.

johnsonm, 2021-06-22:
I stripped out everything I could (had to leave currently-unused Z axis config in place) and checked all pins. And yes, safety first. Not even hooking up the laser until I've tested control for it, and yes of course don't depend on code or config for safety. Thanks for confirming file contents!

PR #4537 (2021-09-03): Add referenced config example
This  PR adds the referenced PWM-Tool config example, which closes #4410.
closed 2021-09-03 by KevinOConnor