Klipper3d/klipper #4419: Accelerometer diagnostics

onfoot, 2021-06-24:
This is more like a feature idea/request/sanity check.

Since we can mount accelerometers on some types of printers permanently, we can leverage them to perform different types of diagnostics, like belt tension, if there is any backlash due to for loose grub screws or lead screw nut being worn out, bent extrusions, rails, rails binding up, whether motion is smooth in most directions etc.

Are there any experiments done by someone on this or any desire to add such features to Klipper?

Sineos, 2021-06-24:
https://klipper.discourse.group/t/precise-belt-tension-for-corexy/367

Please direct such discussions to Discourse

onfoot, 2021-06-25:
Thanks, will do!

closed 2021-06-25 by onfoot