Klipper3d/klipper #4427: Bed temp overshoots and never recovers

RafaelSchridi, 2021-06-26:
Me and some other users on the Artillery Discord have been trying to solve the problem of the bed temperature overshooting and basically "soft locking" the print because it never goes back down.

(manually stopped here)
[image]

When I use the PID values I still have of my marlin installation it also overshoots but swiftly recovers.
[image]

The values produced by each PID tune are also VERY different as seen here.
[image]

tanaes, 2021-06-26:
Chiming in, this seems to be a pretty common problem among users on the Artillery Discord, and in most cases, replacing the Klipper-tuned PID values with their previous Marlin-tuned values gets rid of the problem.

At a bit of a loss as to why this might be. The only meaningful difference I can find in the code between the two tuning loops is that the Marlin code has a Kp adjustment factor that changes between hotend (0.6) and bed (0.2) tuning, where Klipper seems to use 0.6 for all. (Caveat: I am easily lost in C code!)

If the problem really is an overestimated Kp due to asymmetric temperature response (as suggested in the aforementioned PR), is it possible that this (presumably empirically-derived) correction is worth implementing?

I'll try to get some other users who've experienced this to contribute data, and if anyone has a suggestion as to a preferred style of solution I can try whipping up a branch with it and have folks test it out.

KevinOConnor, 2021-06-27:
Interesting.  Klipper uses the basic Ziegler–Nichols "classic PID" (see https://en.wikipedia.org/wiki/Ziegler%E2%80%93Nichols_method).  It looks like in 2019 Marlin changed the bed calibration to use Ziegler–Nichols "no overshoot" (Marlin commit 1db7013e).

I'd be reluctant to hard-code the bed to use a different algorithm, but I can see where using "no overshoot" would be useful for heaters that heat much faster than they cool.  So, I agree Klipper could be improved here.

For testing purposes, it's easy to calculate the "no overshoot" PID from the "classic PID" parameters - just divide the recommended Kp by 3, divide Ki by 3, and multiply Kd by 0.88 .

-Kevin

tanaes, 2021-06-27:
I agree that hard-coding the bed behavior doesn't sound great. 

Can you think of any issues with discovering the relative heatup/cool down rates from the first cycle and deciding on an algorithm based on that?

tanaes, 2021-06-27:
I'll do some testing on my printer when I have a chance using these two methods.

RafaelSchridi, 2021-06-27:
> For testing purposes, it's easy to calculate the "no overshoot" PID from the "classic PID" parameters - just divide the recommended Kp by 3, divide Ki by 3, and multiply Kd by 0.88 .

I tried doing that with my Klipper PID and it made things considerably worse, although it does slowly recover.
[image]

Also tried what the[ Reprap wiki](https://reprap.org/wiki/PID_Tuning) recommends and also didn't play out well
[image]

Kinda makes me think that there is something wrong with the default values.

KevinOConnor, 2021-06-27:
After some further research, it appears that "no overshoot" is not from Ziegler–Nichols - the wikipedia article is just misleading.  It appears "no overshoot" is from https://www.mstarlabs.com/control/znrule.html which cites "Rule-Based Autotuning Based on Frequency Domain Identification" (https://www.researchgate.net/profile/Keith_Godfrey/publication/3331995_Rule-based_autotuning_based_on_frequency_domain_identification/links/00b7d52e79c2a99e78000000.pdf) which seems to cite "D.  E.  Seborg,  T.  F.  Edgar,  and  D.  A.  Mellichamp,Process Dynamicsand Control.    New  York:  Wiley,  1989.", ...

Anyway, who knows where it originates from or if it is in fact a "standard".

In any case, the other common solution to very slow bed convergence is to change the g-code start commands to not use M190 and instead use `TEMPERATURE_WAIT SENSOR=heater_bed MINIMUM=60`.  That alternative command does not wait for convergence.

-Kevin

jwkicklighter, 2021-07-06:
Just chiming in to say that I'm also experiencing this problem with my Artillery machine. I can provide some graphs from my end later, but in the mean time I just had a question based on the current discussion: would there be any harm in letting the user choose the PID tuning method (standard vs. no overshoot) in the config, or perhaps as a flag in `PID_CALIBRATE` (which would also save the chosen method next to the PID values so Klipper would know which algorithm the was used)? May not be the most desirable thing, but it seems better than hard-coding the no overshoot algorithm flag per specific beds/printers.

robthide37, 2021-07-21:
Try setting smooth time to .1

jwkicklighter, 2021-08-11:
@robthide37 What is this referring to?

robthide37, 2021-08-24:
in the bed heater section . their is a smooth time applied that allows bad oscillation of temps no matter how good the pid values are

bjones14, 2021-09-07:
I have spent some time struggling with PID control in Klipper, and as a result I have started a pull request that adds two new pid algorithms - [no overshoot] as mentioned above, and [ponm] proportional on measurement, which is a much more ideal PID control for integrating (heating) processes like this - http://brettbeauregard.com/blog/2017/06/introducing-proportional-on-measurement/

tanaes, 2021-09-07:
Oooh that looks *very* promising. Thanks for taking it on!

robthide37, 2021-09-07:
You really should try the smooth time before spending alot of time cause you will see default is like 2 sec iirc and this just always has it chasing the end of an oscillation. It's why values that marlin and rrf can use with no prob will not work on klipper sometimes to point the print cannot start but with the smooth time set to 10 ms or around that it is rock solid temp and I can heat to 110 and not even hit 111 on warmup like it will hit 110.6 then settle at 110.0 +-.1 and maybe a +-.5 when fan kicks on

bjones14, 2021-09-07:
> You really should try the smooth time before spending alot of time cause you will see default is like 2 sec iirc and this just always has it chasing the end of an oscillation. It's why values that [...]

You are absolutely correct, though I do believe the smooth time default was at 2 seconds and now it defaults to 1 second.  A long smooth time delays the response of the PID which can cause control oscillation.  

In my case, the additional PID algorithms I added were for extruder tuning, as I am unable to keep a hotend at 250C and 100% layer cooling fan in Klipper (loses control when fan kicks on and never returns) - the classic Z-N PID isn't adequate for the disturbance that the layer fan induces with those settings, and without implementing a feed forward in the PID for the layer fan speed, there isn't a great way to overcome this limitation.

RafaelSchridi, 2021-09-07:
Also setting smoothing to .1 sends my bed towards max temps
[image]

EuanWebster, 2021-09-09:
This has been driving me nuts for months on my Artillery and my Tevo. Granted, I did stop using it for a while so I could actually print on the printer. I've just put the PID values in manually (P45.20 I7.46 D182.66, from https://gist.github.com/vogler/588c577a37f5a573afa5b1000307d41b ), which I believe are close to stock Marlin figures. Worked perfectly first time. The bed is stable within 0.5 degree. These figures for I & D are very far away from the Klipper figures that I get from a PID tune. 

Both of the above printers have 230V beds controlled by SSRs. Both are bed slingers. Absolutely no idea if this is relevant, but Klipper bed PID tune on my Vorons works beautifully.

robthide37, 2021-09-11:
redo the the pid tune after then try

RafaelSchridi, 2021-10-17:
I think this is waiting on #4667

watzon, 2021-11-22:
This is definitely still an issue and it would be nice to have it resolved

EuanWebster, 2021-11-23:
I'm having to go back to Marlin on two printers to resolve this issue. I see other people perplexed by it in various forums several times a week.

jwkicklighter, 2021-11-30:
I've had to use the gcode band-aid to have the printer not "wait" for the bed temp, which means that the bed is trying to hit the right temp and overshooting for the whole print.

robthide37, 2021-12-01:
u tried smooth time.1 ?

RafaelSchridi, 2021-12-05:
I recently moved to an alu plate bed and needed to recalibrate so I did the PID tune with .1 smoothing and it's gone off the walls.

[image]

So is my only solution here to just accept an overshoot or reflash marlin to PID tune there?

RafaelSchridi, 2021-12-05:
Heres my heattest.txt from WRITE_FILE=1 if that is of any use:
[heattest.txt](https://github.com/Klipper3d/klipper/files/7655027/heattest.txt)

alvarini, 2021-12-28:
Same problem here with the bed of an artillery X1 (and also the hotend is not so stable)
Now I'm playing with stock values but the start printing process delays a lot and makes it very annoying ( I have to wait like 10 minutes each print)
Before
[image]

After with stock values from @EuanWebster 
[image]

julioviegas, 2022-01-08:
This is definitely an issue, as my bed is continuously going +3C above the set temperature.

Also, my hotend is variating +/-3C between the set temperature.

I don't have any changes in my hardware setup, I just upgraded Klipper and that started to happen. I also tried a new PID trune for both bed and hotend and the problem persists. So definitely something is going on here.

robthide37, 2022-01-10:
And you reduced smooth time for the heater to .2 or .1?

DDFranks, 2022-01-21:
Been following this thread, just moved to Klipper on a Sidewinder x1 v4

After PID tuning the glass bed at 55 for PLA, it would over shoot to 57 and hover around there but never lower. 

Changed to the fulament fula-bed re-tuned and same thing.

Added the smooth_time to my config at 10 for the bed, per RafaelSchridi after trying (2, 1, 0.1)

Held for 2hours at 55.3c +-0.1 change and during a print last night.

Thank you for the solution RafaelSchridi

Now I'm working on the PID tune for the extruder tuned for 220, over shoots to 224 hovers then holds at 220 +-1c

Thank you all

Jamin82, 2022-01-27:
We are trying to fix a hardware problem with software here.

The board has a very noisy 5v rail that is creating the sensor noise and making pid tuning difficult

If you add a low esr capacitor to the 5v rail it will smooth out the temperature measurements and everything will work with stock smoothing settings

julioviegas, 2022-01-29:
smooth time 10 for both bed and extruder fixed it for me as well 
+/-1C variation now, amazing!

george4n, 2022-01-30:
I can confirm that setting "smooth_time: 10" for bed and extruder solved the issue with my Artillery Sidewinder X1. I am getting ±0.2°C now

albza, 2022-02-10:
I'm having the same problem but, since I'm not a Artillery owner, I'm adding my experience to this issue.
I have a TwoTrees Sapphire Pro (MKS Robin Nano v1.2) and I've been using Klipper for about a year now and I've always had this problem:
- bed temperature during PID tuning is ok, e.g. if I set it to tune to 40°C it correctly reaches the given temperature 
[image]

- after saving the new PID data with SAVE_CONFIG, if I try to set the exact same bed temperature it overshoots and then stabilizes at 44.5°C 
[image]

I've tried using PID tuning many times but it never managed to solve the issue. 
Honestly I have no idea where to bang my head to :)

Pixelstrudel, 2022-02-11:
By upping the smooth_time to 10 on my X1 v4 I no longer have issues with the bed overshooting however as @Jamin82 mentioned I fear this is not a real solution to the problem.

While I am not entirely sure if this is how the smooth_time parameter operates I noticed that while the output looks fine the noise seems to still be there. This is less obvious (and impactful) with the bed heater but when changing the smooth_time for the hotend I can observe bulging layers with prints. When changing the smooth_time back to the default value this issue is gone. 

So while for the bed this might be fine I will be careful with upping the smooth_time of the hotend. Of course your mileage may vary, especially when considering that I'm not using the stock heat cartridge.

albza, 2022-02-11:
> @albza Have you tried using the `smooth_time: 10` setting like they mentioned above? Curious to see the results with a different printer.

I did try and it helped, but just a little bit. Basically it halved the overshoot but it is still present… at least now it is bearable!

nstetter, 2022-02-12:
> 
Could you elaborate which exact capacitor this would be one would have to swap?

closed 2022-03-27 by github-actions[bot]

Druntar, 2022-04-06:
Just bumping this issue as I recently ran into it myself. Attempting the smooth time fix now. Hope we can get this repaired upstream soon.

ReXT3D, 2022-04-11:
I would be really curious to see if anyone currently experiencing these issues could perform manual PID constant derivation and whether that resolves their issues.

I recently compiled a step-by-step guide that outlines how such manual derivation could be accomplished:

[Manual heater PID constant derivation by example](https://klipper.discourse.group/t/manual-heater-pid-constant-derivation-by-example/2486)](https://klipper.discourse.group/t/manual-heater-pid-constant-derivation-by-example/2486)

ReXT3D, 2022-04-11:
Thank you, just beware that this process takes a long time if you indeed want to attempt it.  I would reserve at least 2-3 hours...

The smooth time only applies to derivative term smoothing in the PID control algorithm.  My approach performs an open-loop manual step test of the heater "system" or process and does not involve the PID control algorithm.  The smooth time value therefore does not affect the process.  In summary, you issue a manual PWM command, let the bed stabilize thermally (loooooong time), then increase the PWM value by 5% and let it stabilize again.  You then use the collected log data to define the first order plus dead time process (FOPDT) parameters or coefficients by using an online curve firing tool.  It will produce the gain, time constant and dead (delay) time.  Those can be then used to define the required PID controller constants by means of many different PID "tuning" techniques.

wernight, 2022-05-02:
I tried `smooth_time: 10` for the extruder and bed and the extruder overshoots by about 5°C (210°C -> 215°C) over about 1 min. Tried `smooth_time: 0.2` and it overshoots by over 20°C (I stopped it before it continued):
[image]

Overall neither are great by *10* is way better for sure.

RafaelSchridi, 2022-05-02:
I don't think it's a great idea to apply smoothing to the extruder. If that's overshooting that might be a separate issue.

wernight, 2022-05-02:
@RafaelSchridi Without `smooth_time` for the extruder (but `smooth_time: 10` for the bed) it still overshoots about 5°C and keeps around 1°C above the desired temperature (not huge but I had to tweak the start script else it may wait a long time before ack that the temperature was reached and stabilized):
[image]

Not sure what I could do to improve here, and it seems that multiple other folks are in the same situation as myself.

jmccaull, 2022-05-27:
I recently switched to klipper from marlin on my v4 X1. I was having the same problems as most in this thread with my extruder, setting smooth_time to 10 is a huge improvement. Without smoothing my extruder overshoots by ~20C and never quite comes down to the specified temp. I'm powering my rpi4 off a buck converter on the printers power supply, could this be contributing? Will experiment with different power supplies over the weekend and report back

MichielioZ, 2022-05-31:
I don't seem to have any of the problems mentioned here, but I am confused about this issue...
So apparently all the people with this issue come from a working Marlin firmware and the difference in code compared to Klipper is known. If implementing the PID from Marlin doesn't solve the issue the problem must be located in the handling of those PID values ? (or is my logic wrong here ?)

EuanWebster, 2022-05-31:
Implementing the PID values from Klipper does help alleviate the issue, but
doesn't remove it totally. Also, I've changed my hotend, so need to PID
tune again. I'm now thinking it's how Klipper runs on the MCU, but I've yet
to swap it out and test again.

RafaelSchridi, 2022-05-31:
The issue seems to have derailed it a bit with people bringing up other problems, the original problem was tuning **_BED_** PID in klipper causes overshooting but using the original marlin values works.

EuanWebster, 2022-05-31:
> 

Yes. It was originally bed, but now is hotend as well. I'm guessing it's all related, but I'll leave it to smarter people than me to confirm that :-D

RafaelSchridi, 2022-05-31:
I think the hotend overshooting is a different issue because the PID tuning algoritms are the same for hot-ends (see: https://github.com/Klipper3d/klipper/issues/4427#issuecomment-869180092), so it's probably better if you open another issue for that.

jmccaull, 2022-05-31:
Sorry I probably should have been more explicit, I was having this issue with both my bed and hotend. The previous marlin numbers did perform better but neither brought the temps down to target. I've since backed off the 10 smooth time down to 3 for both the bed and the extruder and anecdotally I'm now seeing similar performance to what I had with Marlin. I'm consistently seeing 10c overshoot for the hot end and 3c for the bed but they both level out to the target temp within 30s to a minute.

PhilBaz, 2022-07-11:
Smooth of 10 on the bed and 3 on the hotend sorted out all my problems.  Thanks guys! Was having bad overshoots on both, that didn't come down.

Sineos, 2022-08-03:
You might want to test and provide feedback on https://klipper.discourse.group/t/experimental-pid-improvement-changes/3604

kjkent, 2022-11-15:
Just commenting as I too am affected by this issue on an Artillery Genius (basically a small Sidewinder X1 if I'm not mistaken).

Should the issue be reopened? Seems there's active conversation and it's an ongoing problem for folk, but the bot closed it down back in March...

@Sineos I just checked out that link, seems super promising. I'm going to test and offer feedback.

StoneColdCrazy, 2022-11-27:
The problem with the motherboard, which is usually the MKS Gen L V1, is an unstable 5V supply. This fluctuates between 4.7V and 5.1V, depending on the load. I assume a design error here. I see it as rather difficult to find any solution on the part of the software here and also suspect the main problem here why it comes to incorrect measurements. In the logs, you can see the temperatures fluctuate even during the cooling down phase. Sometimes the temperature jumps from measurement to +-0.7°C back and forth. 

Example for the bed:

heater_bed: target=0 temp=59.7 pwm=0.000
heater_bed: target=0 temp=60.2 pwm=0.000
heater_bed: target=0 temp=59.7 pwm=0.000
heater_bed: target=0 temp=59.7 pwm=0.000
heater_bed: target=0 temp=60.2 pwm=0.000
heater_bed: target=0 temp=59.6 pwm=0.000
heater_bed: target=0 temp=59.5 pwm=0.000
heater_bed: target=0 temp=59.4 pwm=0.000
heater_bed: target=0 temp=59.4 pwm=0.000
heater_bed: target=0 temp=59.1 pwm=0.000
heater_bed: target=0 temp=59.2 pwm=0.000
heater_bed: target=0 temp=59.6 pwm=0.000
heater_bed: target=0 temp=59.4 pwm=0.000

jgarder, 2022-12-23:
> I'm having the same problem but, since I'm not a Artillery owner, I'm adding my experience to this issue. I have a TwoTrees Sapphire Pro (MKS Robin Nano v1.2) and I've been using Klipper for about a [...]

i have a sainsmart coreception 300 (same as TwoTrees Sapphire Pro (MKS Robin Nano v1.2)) and i am also experiencing this issue.  adding smoothing :10 seems to have made some difference to my benefit. this is a good band-aid until a real solution is found.

robthide37, 2022-12-23:
I have learned to manuaklly adjust clients as well as increase my smooth
time and I get it to hit target with less than a degree or two overshoot
and no oscillation usually but I am def curious to read up on your method
as i slowly read to get it .

SrgntBallistic, 2023-05-22:
I'm using an MKS Gen L V1 board. I've been having issues with very noisy Thermistor readings from both the bed and the hotend. I've tried multiple hotends, heater cartridges and thermistors in particular on the hotend.

I was also experiencing a lot of overshooting and M190/M109 never completing due to it never dropping enough to steady state around the specified temperature.

Variations I tried:
Hotends
- Stock Tevo hotend
- E3D V6
- E3D Revo
Thermistors
- Stock Tevo (Seemed to work best)
- E3D V6 Stock cartridge thermistor
- WINSINN HT-NTC100K Generic 3950
- Fysetc 104NT-4-R025H42G

I tried running a separate thermistor line from all of the rest of the cables to try and account for any types of interference but no luck.

I found a few other threads were ppl noted they had to change their `smooth_time` to a LOWER value. Even as low as 0.1 which I think basically disables it.

Then I came across this thread were it seems like other MKS board users are having to turn their smooth_times way up. I think I recall Kevin mentioning the smooth time particularly helps with noise on the derivative term?

So I've now started upping my smooth time and trying to see the effects.

Here is my smooth_time set to `10` just heating to a desired temp
Thermistor: Fysetc 104NT-4-R025H42G
Hotend: E3D V6
Cartridge: 65W
[image]

A PID Tune with smooth_time `10`
[image]

For PID Tuning I'm still seeing times were it will heat the cartridge at 100% when it's clearly well over the desired temp
----
Here's an example with smooth_time set to the default '1'
[image]

Here's a PID Tuning with that value
[image]
----
Here's a heat up with the value of .1 "off"
[image]

And finally a PID Tuning with that value
[image]

----
So it would seem some/many MKS boards can benefit from a pretty large amount of filtering with a high smooth_time?

I'm not sure if this provides any useful data points. But can we draw any insight about the MKS boards physical heater/thermistor implementation or the firmware for these chips that might help understand why they seem to to maybe be so prone to noise and overshooting with Klipper?

I might play around a bit more with different smooth_time values. Is there anything to worry about with a smooth_time as high as `10`?

SrgntBallistic, 2023-05-25:
Update on the post above:

I added a LowESR 1000uF Capacitor across 5v - Gnd header pings as recommended by a number of ppl with MKS boards and some creality boards that had a lot of noise on their thermistors, and over shoots.

I reset the `smooth_time` back down to 1 and did a new PID Tune. Here are the results:
[image]

klack, 2023-08-05:
I have anAC bed, like the Artillery and `smooth_time: 10` seems to be fixing the problem.

klack, 2023-08-05:
> Update on the post above:
>
> I added a LowESR 1000uF Capacitor across 5v - Gnd header pings as recommended by a number of ppl with MKS boards and some creality boards that had a lot of noise on their [...]

You added it where?

SrgntBallistic, 2023-08-06:
> @SrgntBallistic you added it where?

To an open pin header on the board with 5v and GND next to each other.

It seems the 5v is used as an ADC Vref. So a smoothing cap can help the readings electronically instead of with a firmware setting.

klack, 2023-10-06:
> > @SrgntBallistic you added it where?
> 
> To an open pin header on the board with 5v and GND next to each other.
> 
> It seems the 5v is used as an ADC Vref. So a smoothing cap can help the readings electronically instead of with a firmware setting.

Any way you can show a pic?