Klipper3d/klipper #4428: TMC2208 extruder driver fails

totorus, 2021-06-27:
Hello to all

I'm in the process of setting up klipper on an X5SA with an SKR Turbo V1.4, TMC2208, LDO orbiter extruder and a trianglelabs dragon hotend. 
Everything was working fine with Marlin but with klipper I have a problem with the extruder driver.

I made the settings for the extruder rotation_distance and launched a print of a 20mm cube for the X and Y axis settings.
The print was going well and about halfway through the print I got an error on the extruder driver and the printer stopped.

`TMC 'extruder' reports error: DRV_STATUS: 40190020 s2vsb=1(LowSideShort_B!) CS_ACTUAL=25 stealth=1
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown`

I restarted the printer and after checking that all the connections were good I restarted the print.
The initialization went well but as soon as the printing started the extruder went into error again.

I restarted and then tried to do a simple extrusion and the extruder driver went into error again.

I exchanged it with the Y axis driver and re-run an extrusion test with the same result.
So the driver is good. I replaced it by a new one but the same error again.

I have updated all the software through the Fluidd interface but no change.

I searched everywhere on the internet but I didn't find a clear answer to this concern.
I tried to change the "stealchop_thresold" to different values but without result. Not knowing how to determine this value I tested the following values on various configuration examples found on the net.

Original value: 5
Tested values: 8, 30, 900

The result was always negative.

I'm out of ideas...

I did a "DUMP_TMC stepper=extruder" and here is the result:

"09:32:07  $ DUMP_TMC stepper=extruder
09:32:08  // ========== Write-only registers ==========
09:32:08  // IHOLD_IRUN: 0008190f IHOLD=15 IRUN=25 IHOLDDELAY=8
09:32:08  // TPWMTHRS: 000000d3 TPWMTHRS=211
09:32:08  // TPOWERDOWN: 00000014 TPOWERDOWN=20
09:32:08  // ========== Queried registers ==========
09:32:08  // GCONF: 000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
09:32:08  // GSTAT: 00000002 drv_err=1(ErrorShutdown!)
09:32:08  // IFCNT: 00000013 IFCNT=19
09:32:08  // OTP_READ: 0000000d OTP_FCLKTRIM=13
09:32:08  // IOIN@TMC220x: 20000141 ENN=1 PDN_UART=1 SEL_A=1(TMC220x) VERSION=0x20
09:32:08  // FACTORY_CONF: 0000000d FCLKTRIM=13
09:32:08  // TSTEP: 000fffff TSTEP=1048575
09:32:08  // MSCNT: 000002f8 MSCNT=760
09:32:08  // MSCURACT: 01f50108 CUR_A=-248 CUR_B=-11
09:32:08  // CHOPCONF: 14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
09:32:08  // DRV_STATUS: c00f0000 CS_ACTUAL=15 stealth=1 stst=1
09:32:08  // PWMCONF: c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
09:32:08  // PWM_SCALE: 00000012 PWM_SCALE_SUM=18
09:32:08  // PWM_AUTO: 000e0024 PWM_OFS_AUTO=36 PWM_GRAD_AUTO=14"

Here is the klippy.log file
[klippy.log](https://github.com/KevinOConnor/klipper/files/6721257/klippy.log)

My printer.cfg file
[printer.txt](https://github.com/KevinOConnor/klipper/files/6721275/printer.txt)

I hope you can help me.

vladbabii, 2021-06-27:
> TMC 'extruder' reports error: DRV_STATUS: 40190020 s2vsb=1(LowSideShort_B!)

.. means you have an issue with wires or connector - 2 of the wires of the stepper are touching somehwere and making a short. this is a physical printer issue, not a klipper issue - the tmc driver it's telling you that you have an issue with the stepper motor connection

totorus, 2021-06-27:
Hi @vladbabii  and thanks for your reply.

I just tested a new print by unplugging the motor from the extruder and I no longer get an error message or driver lockup.
It seems that you are right and that I don't know how to read an error message... :)

I will check my wiring and come back here to give the result of my tests.

totorus, 2021-06-27:
Well I didn't find anything strange in my wiring or on my board but still the same problem.

I have another extruder of the same model on the way. I would test with this one and see.

I'm closing the ticket.

closed 2021-06-27 by totorus