Klipper3d/klipper #4433: Update Documentation to use CMD to ssh instead of PuTTY

ChaosBlades, 2021-06-27:
The documentation states....

> If using a Linux or MacOS desktop, then the “ssh” software should already be installed on the desktop. There are free ssh clients available for other desktops (eg, PuTTY). Use the ssh utility to connect to the Raspberry Pi (ssh pi@octopi – password is “raspberry”) and run the following commands:

"ssh" software is already installed on Windows as well and has been for many years. You can simply open CMD and type `ssh pi@octopi.local `just like you can do on Linux/MacOS Terminal. No reason to have the user download additional software when it is not needed.

ChaosBlades, 2021-08-02:
*Activity*

dxgldotorg, 2021-09-03:
Is SSH pre-installed or do you have to add it as an optional feature in Windows 10/11?

charlespick, 2021-09-10:
@dxgldotorg yes it's pre-installed

@ChaosBlades documentation issues almost never go anywhere. Write up the proposed documentation changes yourself and open a pull request. It's a fairly sensible update so my guess is it'll get merged.

closed 2021-12-01 by github-actions[bot]