Klipper3d/klipper #4434: Using the I2C connector on the BTT GTR 1.0 board

DEAth1981, 2021-06-27:
When I try to use the i2c bus (pins PH7 and PH8) on the BTT GTR 1.0 board, I get the error " Unknown i2c_bus 'i2c3' ".
According to the board pinout scheme, PH7=I2C3_SCL, and PH8=I2C3_SDA.
How to use the I2C connector on the board?

P.S. Section code in the board configuration where i2c is used:

[pca9533 led_top]
i2c_address: 98
i2c_mcu: mcu
i2c_bus: i2c3
i2c_speed: 100000
initial_RED: 0
initial_GREEN: 0
initial_BLUE: 0
initial_WHITE: 0

DEAth1981, 2021-06-29:
Since few people are ready to help me here, I decided to duplicate the request for the [Klipper Community Discourse server](https://community.klipper3d.org/)... spent time registering and setting up an account, created a request... and got "Account temporarily on hold". :D

Why this server if you get banned when trying to get help?

Sorry for the offtopic!

KevinOConnor, 2021-06-29:
The Discourse server has various checks for spammers - your message was only temporarily delayed.  That said, Discourse is the preferred location for topics like this.

-Kevin

DEAth1981, 2021-06-30:
@KevinOConnor, my request is not related to configuration, but to full i2c support on STM32F4xx in Klipper. Due to the fact that when compiling the firmware for BTT GTR 1.0 (STM32F407) in terms of the i2c implementation, the code from STM32F1xx is used (and this chip does not have i2c3), then the i2c3 implementation is completely absent in the compiled firmware for BTT GTR 1.0 (STM32F407).

And since I do not understand anything in programming, I made a request with the hope of knowledgeable people who can bring this implementation (i2c3 support) to life.

So I don't quite understand why there is a tag "not on github"?

closed 2021-07-02 by github-actions[bot]