Klipper3d/klipper #4435: Update Klipper to use Python 3

ChaosBlades, 2021-06-27:
I can't seem to find an issue closed or otherwise to follow progress or discussions regarding Klippers use of deprecated Python 2.

> DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.

Can we get an update on the progress of this migration or a confirmation of 'won't fix'?

jakep82, 2021-06-28:
https://github.com/KevinOConnor/klipper/pull/3278

ChaosBlades, 2021-06-28:
Github search is dumb.

closed 2021-06-28 by ChaosBlades