Klipper3d/klipper #4436: Unable to obtain 'trsync_state' response

hesahinzaxe, 2021-06-28:
I am using a triple z system. When I use a probe as an endstop, everything is fine. However, if I configure 3 endstops (mechanical switches) and send z_tilt_adjust, it goes to the shutdown state and raises the errors '!! Unable to obtain 'trsync_state' response' and 'Timer too close'. Only change in the config is the endstop configuration. 

Here is my log files
Probe as the endstop: [klippy.log](https://github.com/KevinOConnor/klipper/files/6724314/klippy.log)

Three endstops: [klippy.log](https://github.com/KevinOConnor/klipper/files/6724321/klippy.log)

hesahinzaxe, 2021-06-28:
I have noticed that if I move the bed up with G1 Z5 after homing, z tilt can be succeeded. If I send the command at the homing position (position max), it cannot tilt.

KevinOConnor, 2021-06-28:
FYI, it's generally preferable to report on Klipper Discord or Discourse first.  That said, I looked at your log and it does appear Klipper is not operating correctly.  I'll investigate further; if you have a workaround you'll probably want to use that until it can be tracked down.

-Kevin

goosst, 2021-06-28:
I've upgrade to TMC2226 stepper drivers and updated to the latest klipper firmware --> started having identical issues. 
(Actually I only updated klipper to use this rotation_distance setting and be able to follow the documentation.)

I'm using a delta printer, so homing three axis at the same time with (mechanical) switches as stop. Seems like it's not a new issue but I've attached my log file in case it helps with anything.

Thanks for the great firmware but I might suggest to create more tags / releases of stable versions, so the average user isn't by default pointed to a trunk.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/6728862/klippy.zip)

DEAth1981, 2021-06-28:
A similar problem... two optical limit switches on the Z axis (parking in Zmax), there is also a BLTouch
[klippy.log](https://github.com/KevinOConnor/klipper/files/6729104/klippy.log)
... when I try to execute SCREWS_TILT_CALCULATE or Z_TILT_ADJUST, I get the printer stop with the error "Unable to obtain 'trsync_state' response"

hesahinzaxe, 2021-06-29:
> 
> 
> FYI, it's generally preferable to report on Klipper Discord or Discourse. That said, I looked at your log and it does appear Klipper is not operating correctly. I'll investigate further; if you have a workaround you'll probably want to use that until it can be tracked down.
> 
> -Kevin

I have reported it on discord. Nobody could help. Then I opened the issue. I have a workaround and I will be waiting the fix. Thank you Kevin.

hesahinzaxe, 2021-06-29:
I have tried to use the macro below:

[gcode_macro G30]
gcode:
        G1 Z5
	Z_TILT_ADJUST

It still raises the error. However, if I separately send these two command, it does not raise it. Interesting.

KevinOConnor, 2021-06-29:
Alas, on closer inspection, the provided log does not appear to be a complete log file (it is missing version information).  Unfortunately, without a full log there's nothing I can really do to help.

If you can reproduce this issue, please open a new topic on the Klipper Discourse server and attach the full log from the event (zip the file if it is large).  Also, make sure you are on the latest code from this repo (no code changes or external code at all).  To be sure, you can ssh into the rpi and run something like (`cd ~/klipper ; git fetch ; git checkout origin/master`).

-Kevin

Digital-apprentice, 2021-07-07:
Hi guys, I had the same problem, I was getting the same error. I solved it by increasing the speed in the quad_gantry_level (Z_TILT_ZDJUST) section and in the probe section. After this correction, the error no longer appears.

KevinOConnor, 2021-07-09:
As mentioned on Klipper Discourse, the underlying issue appears to be due to a 32bit timer rollover issue introduced by an omission in the new trsync homing code.  Hopefully it is now fixed (commit f2421005).

-Kevin

DEAth1981, 2021-07-09:
Updated to the latest version of Klipper. If earlier an error occurred during an attempt to execute procedures that involved BLTouch (for example, Z_TILT_ADJUST), now errors associated with TMC drivers are popping up. It became simply impossible to print.
[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/6793492/klippy.log)

screech24, 2021-07-14:
Still getting Errors on my delta after updating to the latest commits maybe its unrelated but I get " Unable to read tmc uart 'stepper_c' register GSTAT" as well as " Unable to obtain 'trsync_state' response" shown in the console upon shutdown..
attached is my log. 
[klippy-2-8.log](https://github.com/KevinOConnor/klipper/files/6819451/klippy-2-8.log)

closed 2021-07-22 by github-actions[bot]