Klipper3d/klipper #4448: Unable to install klipper Ender max 3 4.2.7 board

Takohtuesdays, 2021-07-01:
[klippy.log](https://github.com/KevinOConnor/klipper/files/6751295/klippy.log)

This happens when I try to make flash file:

make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
  Flashing out/klipper.elf.hex to /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 via avrdude
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x42
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x32
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x2e
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x39
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x37
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x20

avrdude done.  Thank you.

make: *** [src/avr/Makefile:34: flash] Error 1

aepkolol, 2021-07-04:
You flash the 4.2.7 board by putting the .bin file it creates on the root of a microSD card and inserting it into the ender. You can't flash that board by USB.

Takohtuesdays, 2021-07-04:
@aepkolol yea I flashed the .bin file via SD card. But I’m trying to install klipper to octoprint

Takohtuesdays, 2021-07-04:
Via ssh

aepkolol, 2021-07-04:
The log you pasted above is avrdude failing to connect to a USB device (since you can't flash the 4.2.7 board)

You don't install Klipper to octoprint, octoprint is used to remotely monitor and control the printer. You just need to config octoprint to connect to Klipper via a virtual serial connection. 

See: **Configuring OctoPrint to use Klipper** at the link below

https://www.klipper3d.org/Installation.html

closed 2021-07-05 by github-actions[bot]