Klipper3d/klipper #4452: I'm having an issue with my bed temp. Keeps failing to heat when it gets to say 50/60 (I'm doing tests at up to 100 mainly). I've been told the issue is around the 2 degrees per 60 seconds seconds with a +/-5 of the target. I've attempted to change this in my printer.cfg but sadly this just gives me further error codes🤷🏻‍♂️  I'm using a Bigtreetech octopus V1.1 with 2209s and an OEM CR-10 S5.  Please feel free to ask any further questions sorry if I've missed key information.  Thanks in advance for your help!

batmanreturns1, 2021-07-03:


nonchip, 2021-07-03:
and you really see nothing wrong with this issue?

batmanreturns1, 2021-07-03:
> and you really see nothing wrong with this issue?

Didn't really come here for sarcastic remarks. I came here to see if there was a way in editing the printer.cfg. To enable me to use the printer. Considering this is a common issue with the CR10S5 with most users upgrading their heated beds because of it. So no I don't see the issue here and apparently neither do many other users. I have attempted to edit this in my printer.CFG as prompted to in the config file with nothing but error messages.

nonchip, 2021-07-03:
> Didn't really come here for sarcastic remarks.

well given you replied like that instead of fixing anything about your "ticket", i guess you really didn't get it. one example: you wrote EVERYTHING in the subject, left the body empty.

> I came here to see if there was a way in editing the printer.cfg. 

well that's another one, you never showed what you tried.

> Considering this is a common issue with the CR10S5 with most users upgrading their heated beds because of it. 

if it is, i'm sure it's documented somewhere, so what instructions did you try to follow and how

> So no I don't see the issue here

evidently.

> and apparently neither do many other users.

i'm sure they would if they tried to read that giant blurb you call a subject line.

> I have attempted to edit this in my printer.CFG

how?

> as prompted to in the config file

where?

> with nothing but error messages.

which?

---

so tl;dr:
* a ticket should have a short subject and details in the actual message so people can actually read it
* you should tell people *what you did*
* you should tell people *why it said it didn't work*
* your "ticket" is the equivalent of getting an email with no body and in the subject line it says "my printer doesn't work despite there being paper in it! why does it not work? i put paper in there! it says something on the display but i'm not telling you what! now that your eyes are already bleeding from this all being one giant bold line, let me add 5 more sentences!"

KevinOConnor, 2021-07-04:
@nonchip - your message is off-topic here.  The original poster was informed that this is not the right place to look for help.  It is not necessary, nor appropriate, to berate them.  I ask that everyone keep github messages on-topic and that they maintain a positive tone.

-Kevin

closed 2021-07-05 by github-actions[bot]