Klipper3d/klipper #4462: Atmega328 pins PE2 and PE3 not supported?

brandon3055, 2021-07-05:
I'm not sure if this would be considered a bug because these pins are not normally used in standard Arduinos. But these are sometimes used as additional analog input pins (A6 and A7)
In my case i am in the process of designing an atmega328 based klipper expansion board and i would really like to use these pins for thermistor inputs.  
If this isn't a simple fix then no problem. I can find a work around. I just want to know if there is any chance we could get support for these pins?

brandon3055, 2021-07-06:
So i decided to fiddle with klipper and see if i could get these pins working. 
I made what i thought were the required changes to pins.py and adc.c

https://github.com/KevinOConnor/klipper/blob/master/klippy/pins.py
```
Arduino_analog_standard = [
    "PC0", "PC1", "PC2", "PC3", "PC4", "PC5", "PE0", "PE1", "PE2", "PE3",
]
```

https://github.com/KevinOConnor/klipper/blob/e7b0e7b43bbf20bf89f47444fbbfc0e10aca1ed1/src/avr/adc.c
```
#if CONFIG_MACH_atmega168 || CONFIG_MACH_atmega328 || CONFIG_MACH_atmega328p
    GPIO('C', 0), GPIO('C', 1), GPIO('C', 2), GPIO('C', 3),
    GPIO('C', 4), GPIO('C', 5), GPIO('E', 0), GPIO('E', 1), GPIO('E', 2), GPIO('E', 3),
```
```
#if CONFIG_MACH_atmega168 || CONFIG_MACH_atmega328 || CONFIG_MACH_atmega328p
DECL_ENUMERATION_RANGE("pin", "PE0", GPIO('E', 0), 4);
#endif
```

I was able to get klipper to accept the pins but u couldnt actually get them to work. They just read as floating analog pins. 
So i'm not sure if maybe i missed something somewhere? Or maybe its just not that simple to add these pins.

KevinOConnor, 2021-07-07:
Best to ask technical questions like this on Klipper Discourse, as we don't use github for questions.

-Kevin

closed 2021-07-08 by github-actions[bot]