Klipper3d/klipper #4469: BTT SKR Pro 1.2 and BTT TMC2209 UART: Unable to read tmc uart 'stepper_y' register IFCNT

m3bolt, 2021-07-07:
Hello, 

I have a strange issue with my 2 X carriages and Y carriage, I am unable to get past this point in the configuration and printer setup. 

I seem to have some sort of communication issue with my TMC2209 UART drivers and cannot for the life of me figure this out. I even went as far as to buy BTT TMC2209s thinking it would solve the issue and it hasn't. 

When I hit home on any axis, klippy shuts down because of a communication timeout error with the TMC drivers. I assumed that this was a UART pin configuration issue, but I've double and triple checked every pin, tried swapping TX and RX pins, tried specifying TX pins, everything. My UART jumper is set correctly on the SKR Pro 1.2 board and my pins are configured using the SKR Pro 1.2 pinout file from BTT's GitHub. 

After I do a firmware restart, the console shows "Unable to read tmc uart 'stepper_y' register IFCNT"

I just can't figure this out and am hoping it's just a bug with some simple solution. I have no jumpers on MS1,2,3 pins because the BTT Manual says not to have any jumpers apart from the UART one. 

Thank you for any help, attached are my log and config. Do note that the Y (and X2) steppers are in E1 and E0 stepper driver spots, but the UART pins are still properly configured for E1 and E0 stepper driver spots.

[M3BoltConfig.txt](https://github.com/KevinOConnor/klipper/files/6780242/M3BoltConfig.txt)
[m3bolt_klippy_log.txt](https://github.com/KevinOConnor/klipper/files/6780243/m3bolt_klippy_log.txt)

m3bolt, 2021-07-07:
If I configure my Y driver according to [this website](https://www.gitmemory.com/issue/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/187/759833334) I get a different error. 

The config from that site is:
`[tmc2209 stepper_x]
uart_address: 0
uart_pin: PC13
run_current: 0.760
hold_current: 0.500
stealthchop_threshold: 250
diag_pin: PB10
driver_SGTHRS: 100`

Which doesn't throw the IFCNT error, but instead throws the `Timeout on wait for 'tmcuart_response' response` error when I hit home, and obviously causes Klippy to shutdown

m3bolt, 2021-07-08:
So where do you actually get help with a Klipper issue? Obviously this is something related to Klipper as Marlin is able to address these UART pins just fine. The Discord is rarely ever a help for actual issues like this due to its fast pace messaging vs an organized forum. Maybe Klippers on Reddit? I get not wanting to flood github with people's questions but this is obviously more than just a simple pin configuration or bad drivers.

joris57, 2021-07-08:
Maybe the Discourse forum? First option on the contact page.

closed 2021-07-10 by github-actions[bot]