Klipper3d/klipper #4473: interacting with SD Card menu item on LCD whlst printing causes printer to crash

strayr, 2021-07-10:
Whilst printing if i touch the SD Card menu item the LCD menu I get an unhandled exception. 

This trips my relay in moonraker.conf and powers down everything, so the failsafe relay works but I'm not sure if I'm capturing everything required. I can rewire if you need more info.

I think the default pause/resume menu items are supposed to be in the SDCard section? As a workaround I've got versions in Filament.

If the problem is that displaying virtual SD card files is not allowed and _won't ever work_ when printing, perhaps we can solve this by not displaying the SD Card contents when printing and having print control appear elsewhere? I think I can see how I'd sort that so if that's the right direction I can try sorting that?

[klippy (1).log](https://github.com/KevinOConnor/klipper/files/6794423/klippy.1.log)

KevinOConnor, 2021-07-11:
This is not normal behaviour.  However, it will take someone to diagnose the situation on your particular printer, and that's not something we use github for.

-Kevin

bcmob, 2021-07-12:
A similar problem.
Also disappeared from the list of buttons "Start Printing", "Pause", "Cancel"

closed 2021-07-14 by github-actions[bot]

bcmob, 2021-07-17:
Cannot start printing from the screen. Klipper falls
[klippy.log](https://github.com/KevinOConnor/klipper/files/6834997/klippy.log)