Klipper3d/klipper #4478: Allow negative unretract_extra_length to prevent oozing

marcocipriani01, 2021-07-10:
I would like to set a negative `unretract_extra_length` (-0.03mm) for firmware retraction in order to prevent oozing from the nozzle, but Klipper doesn't let me do that. Would it be possible to remove this limit? Both Cura and PrusaSlicer allow it without firmware retraction.
Kind regards,
Marco Cipriani

marcocipriani01, 2021-07-11:
I am sorry, this is not a help or feature request. I've also sent a message on Discord but got no answer. I'm asking, if possible, to change the minimum value of `unretract_extra_length` in [klippy/extras/firmware_retraction.py](https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/firmware_retraction.py) to, for exaple, -100:

        self.unretract_extra_length = config.getfloat(
            'unretract_extra_length', 0., minval=something_negative)

This is because of oozing issues with my extruder.
Kind regards,
Marco Cipriani

dawidmpunkt, 2021-07-12:
@marcocipriani01 Does the change work for you? You can change it in your firmware_retraction.py file. Delete the firmware_retraction.pyc file and restart the raspberry pi. Then you can report, if your print improved.

marcocipriani01, 2021-07-12:
Klipper accepts the new minimum value and no longer throws an error. I still haven't printed with this change to the Python code (I'm pretty sure it works), but I can say that stringing improved when I set that value in the slicer's settings.

dawidmpunkt, 2021-07-13:
Alternatively: Does tuning pressure advance solve the issue with oozing? If it does not, you could maybe submit a pull request.

marcocipriani01, 2021-07-13:
I do have pressure advance already enabled. Anyway, this is very a small detail, I'll make a PR and see if it gets accepted. Otherwise I'll stick with slicer retraction.

closed 2021-07-15 by github-actions[bot]