Klipper3d/klipper #4479: SKR2 USB/PSU Power jumper with Klipper not working

anirudh19oct, 2021-07-11:
Hello! I just set up Octo and Klipper on the SKR2. It prints fine and works well **if** the SKR2 board is powered using the USB jumper. If it is switched to PSU power, Klipper doesn't run even though the board powers up and BLtouch does its self-test. I also tried to mask the 5V pin on the USB to just send in data from the Pi to the board but still the same issue.  

Please help. I wish to use PSU power for the board.

dawidmpunkt, 2021-07-12:
This could be a problem with your board. My SKR 2 works fine if I switch it from USB to VDD.

anirudh19oct, 2021-07-13:
> This could be a problem with your board. The 5V voltage regulator might be dead. My SKR 2 works fine if I switch it from USB to VDD.

Funny thing is, it all works fine if I use a USB hub in between Pi and printer.

dawidmpunkt, 2021-07-13:
Sounds nasty. Have you tried another USB cable? If it does not work, I would suggest to wire SKR2 and rPi via UART instead of USB.

dawidmpunkt, 2021-07-13:
> Funny thing is, it all works fine if I use a USB hub in between Pi and printer.

Do I understand correctly: Jumper is set to VDD. If you connect the board to the Pi via USB Hub, everything works? Then there is an issue with the cable.

closed 2021-07-15 by github-actions[bot]