Klipper3d/klipper #4487: samd_sercom section seems to apply to all MCUs

colinfindlay-nz, 2021-07-12:
Configuration attached.

I have 3 MCUs currently configured, an LPC Smoothieboard as the main controller, an ATMega32U4 & recently a SAMD21
I'm attempting to configure the SAMD21 board to use SPI to talk to an AXDL345 - but this requires setting of a `samd_sercom` config block. I'm 90% sure I have the config correct (Although the pins might be slightly out) - however, Klipper is complaining that the MCU doesn't support `sercom_set_pin`
Reviewing the logs and the code - it looks like Klipper tries to apply the SERCOM config to all MCUs - and fails on the first one (Smoothieware)

Logs attached.

FHeilmann, 2021-07-13:
you need to preface your `sercom` parameter with your MCU (`sercom: sercom0` becomes `sercom: adxl:sercom0`)

colinfindlay-nz, 2021-07-13:
I'll give it a shot. Shall I submit a pull request to improve the docs in
this area? I don't remember seeing anything that would indicate this is
possible?

jakep82, 2021-07-14:
It's explained briefly here. 
 https://www.klipper3d.org/Config_Reference.html#mcu-my_extra_mcu

colinfindlay-nz, 2021-07-15:
The working config is actually somewhere in-between.
@FHeilmann You're correct - we need the mcu specifier- but adding it to the `sercom` declaration results in an invalid bus name error.
@jakep82 You are also correct - the docs do mention adding the mcu name to pins - but my understanding was that the sercom specifier was not a pin. This seems to be true.

The working example I have is (but happy to be proven wrong):

```
[samd_sercom adxl]
sercom: sercom0
tx_pin: adxl:PA5
rx_pin: adxl:PA6
clk_pin: adxl:PA7
```

closed 2021-07-15 by colinfindlay-nz