Klipper3d/klipper #4490: Klipper firmware killed my SKR board

kaito83, 2021-07-14:
I made the Robin_nano35.bin and copied to a freshly formated SD card and flashed it, after 100% update wont do nothing, waited about 5min, restarted the printer and it just says booting, I double checked everything before I did this!
Here is my make settings
[image]
How Can I repair the board?

Angelol125, 2021-07-15:
If you title your post as "klipper killed my skr board" and then you proceed to say that you flashed your board with firmware intended to be used with the robin nano board, then you clearly killed your board, not klipper.

kaito83, 2021-07-15:
First my tittle is: Klipper firmware killed my SKR....
Second: make flash cannot upload FW to the board so I have to flash via SD card
Third: I did everything as described, check my picture, check the description

Angelol125, 2021-07-15:
> First my tittle is: Klipper firmware killed my SKR....

That is also the exact same thing I said, isn't it?

> Third: I did everything as described, check my picture, check the description

I saw your screenshot. But I am still unable to understand whether you have an SKR board or a MKS Robin one. And the reason I am unable to understand that, is because in the title you say you have an SKR board in contrast to the commands you try to run, which are meant for an MKS Robin. So in order for anyone to be able to help you, why don't you give any more details, like the name of the board you are trying to flash klipper with?

kaito83, 2021-07-15:
Okay fine now a realized the mistake so my board is a MKS Robin Nano v1.2

Angelol125, 2021-07-15:
Right.. So, in which format have you formatted your SD card? FAT32? Also, what is the capacity of that SD card?

kaito83, 2021-07-15:
Fat32, I guess windows default, 4gb, the tft shows only booting...

Angelol125, 2021-07-15:
if you open up the sd card from your computer, what files are there? is there a firmware.cur file?

Angelol125, 2021-07-15:
Also, you need to give some more info. Like how are you trying to connect with klipper? Octoprint, Mainsail, Fluidd? What is your printer.cfg? Uploading the full log too. Can't work with just that smidge of info..

kaito83, 2021-07-15:
Octoprint
Cfg file
# This file contains common pin mappings for the Two Trees Sapphire
# Pro printer from 2020. To use this config, the firmware should be
# compiled for the STM32F103. When running "make menuconfig", enable
# "extra low-level configuration setup", select the 28KiB bootloader,
# serial (on USART3 PB11/PB10) communication, and set "GPIO pins to
# set at micro-controller startup" to "!PC6,!PD13".

# Note that the "make flash" command does not work with the Sapphire
# Pro. After running "make", run the following command:
#   ./scripts/update_mks_robin.py out/klipper.bin out/Robin_nano35.bin
# Copy the file out/Robin_nano35.bin to an SD card and then restart the
# printer with that SD card.

# See docs/Config_Reference.md for a description of parameters.

[stepper_x]
step_pin: PE3
dir_pin: !PE2
enable_pin: !PE4
microsteps: 16
rotation_distance: 32
endstop_pin: !PA15
position_endstop: 0
position_max: 230
homing_speed: 50

[stepper_y]
step_pin: PE0
dir_pin: !PB9
enable_pin: !PE1
microsteps: 16
rotation_distance: 32
endstop_pin: !PA12
position_endstop: 230
position_max: 230
homing_speed: 50

[stepper_z]
step_pin: PB5
dir_pin: PB4
enable_pin: !PB8
microsteps: 16
rotation_distance: 8
endstop_pin: !PA11
position_endstop: 0.5
position_max: 230

[extruder]
step_pin: PD6
dir_pin: !PD3
enable_pin: !PB3
microsteps: 16
rotation_distance: 6.720
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC3
sensor_type: ATC Semitec 104GT-2
sensor_pin: PC1
control: pid
pid_Kp: 14.669
pid_Ki: 0.572
pid_Kd: 94.068
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PA0
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC0
control: pid
pid_Kp: 325.10
pid_Ki: 63.35
pid_Kd: 417.10
min_temp: 0
max_temp: 130

[fan]
pin: PB1

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command

[printer]
kinematics: corexy
max_velocity: 250
max_accel: 4500
max_z_velocity: 25
max_z_accel: 100

[static_digital_output reset_display]
pins: !PC6, !PD13

kaito83, 2021-07-15:
Yes and is there a firmware.cur

Angelol125, 2021-07-15:
Right..  Can you try adding 

baud: 250000

to your [mcu] section? It should look like this:

```
[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command
baud: 250000

```
Also, if you run `ls /dev/serial/by-id/*` on ssh, what does it return?

closed 2021-07-17 by github-actions[bot]