Klipper3d/klipper #4494: BED_SCREWS_ADJUST Does Not Respect Z Offset

DeanCording, 2021-07-16:
I have a number of different build plates that require different Z offsets to accommodate differing heights.  When I do a manual bed level using BED_SCREWS_ADJUST, the head movement to each screw position does not respect the current Z offset.  As a result, the nozzle crashes into the build plate.

Looking at the code, bed_screws.py calls manual_move in toolhead.py directly, and so moves to an absolute postion. 

In comparison, moves made in gcode_move.py add the gcode offset to the destination position before calling move in toolhead.py.

A workaround is to create a gcode macro for each build plate that changes both the gcode offset and z-probe heights for each.

blalor, 2021-07-28:
Adjusting the bed screws sets the reference point from which the Z offset is applied.  Have you tried leveling the bed using the thickest sheet and then applying `SET_GCODE_OFFSET` in your start g-code?  I've worked up a system for choosing sheets and calibrating their associated offsets via the display; when I change from a smooth to textured plate, I select the new plate from the display and the saved offset is applied in the start g-code. I also have a fairly quick first-layer calibration that I run to find the right squish.

If you're using different build *plates* -- rather than magnetic sheets or similar -- the above doesn't apply. 🙂

DeanCording, 2021-07-30:
The build plates are sheets of glass or other material that sit on top of the printer's bed.  My Z end stop is at 0.
I have a similar method of setting offsets but use Mainsail's presets to set the offset and save it in a stored variable, along with a delayed g-code macro to reload that offset on startup.

The main issue is that it is possible to cause a nozzle crash by using the g-code `BED_SCREWS_ADJUST` when a build plate is in place.

DeanCording, 2021-09-05:
Poke

KevinOConnor, 2021-09-21:
I'm not sure what is being reported here.  If you're looking for help, you'll need to raise on Klipper Discord or Discourse.

-Kevin

DeanCording, 2021-09-24:
The issue is that you cannot use the command `BED_SCREWS_ADJUST` on a printer that has swappable print beds with different thicknesses.  This is because it calls `toolhead.manual_move` which works in absolute coordinates.

In my case, I use a glass plate on top of the heated print bed for PLA, and kapton tape on top of the heated print bed for ABS.  Both of these have different thicknesses which I account for by setting appropriate Z offsets.  `BED_SCREWS_ADJUST` ignores this offset and so can result in a print head crash if I try to level with the glass plate installed.

`BED_MESH_CALIBRATE` has the same issue, so I cannot create a bed mesh for different print beds.

One solution would be to allow `probe_height` to be changed in GCode so that it can be offset to account for the thickness of different build plates.

KevinOConnor, 2021-09-24:
That is the intended behavior of both the the bed_screws and bed_mesh modules.

If the physical printer is changed it is currently necessary to also modify the printer.cfg file.  Some people will make config snippets and change which snippets they include when making changes like this.

-Kevin

closed 2021-11-06 by github-actions[bot]