Klipper3d/klipper #4501: Klipper bug or problem with settings

manroop5, 2021-07-20:
I recently converted my self made cartesian from octoprint to klipper. Overall its better however i believe some bugs in klipper or fluidd. 
[image]
my printer has dead space on all sides. Also my bltouch offset is -48 on x axis. 
The print starts off center to the right. When bed probing set to mesh_min = 0,30  the probing starts at x=48, y30 and the third probe point falls off the edge of bed and error is issued/printer halts. If I change bltouch offset to -20 on x axis it does work but the print starts off the center. 
Gcode preview is hence messed up as well and displays the object off the bed as attached in image
[image]
 Read everywhere. No answer so far. 

config file 
[include kiauh_macros.cfg]
# This file contains common pin mappings for the BigTreeTech SKR PRO.
# To use this config, the firmware should be compiled for the
# STM32F407 with a "32KiB bootloader".

# The "make flash" command does not work on the SKR PRO. Instead,
# after running "make", copy the generated "out/klipper.bin" file to a
# file named "firmware.bin" on an SD card and then restart the SKR PRO
# with that SD card.

# See docs/Config_Reference.md for a description of parameters.

[stepper_x]
step_pin: PE9
dir_pin: PF1
enable_pin: !PF2
microsteps: 32
rotation_distance: 40
full_steps_per_rotation: 200
endstop_pin: tmc2209_stepper_x:virtual_endstop
position_min: -112
position_endstop:  -112
position_max: 200
homing_speed: 150
homing_retract_dist: 0

[stepper_y]
step_pin: PE11
dir_pin: PE8
enable_pin: !PD7
microsteps: 32
rotation_distance: 40
full_steps_per_rotation: 200
endstop_pin: tmc2209_stepper_y:virtual_endstop
position_min: -50
position_endstop: -50
position_max: 200
homing_speed: 150
homing_retract_dist: 0

[stepper_z]
step_pin: PE13
dir_pin: !PC2
enable_pin: !PC0
microsteps: 32
rotation_distance: 8
full_steps_per_rotation: 200
endstop_pin: probe: z_virtual_endstop
homing_speed: 150
position_min: -3
position_max: 250
#position_endstop: 250

[extruder]
step_pin: PE14
dir_pin: PA0
enable_pin: !PC3
microsteps: 32
rotation_distance: 32.83
full_steps_per_rotation: 200
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PB1 # Heat0
sensor_pin:  PF4 # T1 Header
sensor_type: ATC Semitec 104GT-2
control: pid
pid_Kp: 36.162
pid_Ki: 1.929
pid_Kd: 169.510
min_temp: 0
max_temp: 280
pressure_advance: 0.60
pressure_advance_smooth_time: 0.040000
min_extrude_temp: 60
max_extrude_only_velocity: 120
max_extrude_only_accel: 1500

#[extruder1]
#step_pin: P0.1
#dir_pin: P0.0
#enable_pin: !P0.10
#heater_pin: P2.4
#sensor_pin: P0.25
#...

[heater_bed]
heater_pin: PD12
sensor_pin: PF3 # T0
sensor_type: EPCOS 100K B57560G104F
control: pid
pid_kp = 69.812
pid_ki = 0.905
pid_kd = 1345.636
min_temp = 0
max_temp = 90

[fan]
pin: PC8

[heater_fan extruder_fan]
pin: PE5
heater_temp: 50

#[heater_fan fan2]
#pin: PE6

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f407xx_35004D001051383438373634-if00

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 2000
max_z_velocity: 10
max_z_accel: 800

# Using a Raspberry Pi as a secondary MCU, so we can use GPIOs
[mcu rpi]
serial: /tmp/klipper_host_mcu

[gcode_macro POWER_OFF_PRINTER]
gcode:
  {action_call_remote_method("set_device_power",
                             device="printer",
                             state="off")}

[gcode_macro POWER_OFF_LED]
gcode:
  {action_call_remote_method("set_device_power",
                             device="printer_led",
                             state="off")}

[respond]
default_type: echo
#   Sets the default prefix of the "M118" and "RESPOND" output to one
#   of the following:
#       echo: "echo: " (This is the default)
#       command: "// "
#       error: "!! "
default_prefix: echo:
#   Directly sets the default prefix. If present, this value will
#   override the "default_type".

[idle_timeout]
gcode:
    RESPOND PREFIX="info" MSG="Idle Timeout > Turn off heaters"
    TURN_OFF_HEATERS
    {%if not printer.pause_resume.is_paused %}
      RESPOND PREFIX="info" MSG="Idle Timeout > Motors off"
      M84
      RESPOND PREFIX="info" MSG="Idle Timeout > Printer off"
      POWER_OFF_PRINTER
      RESPOND PREFIX="info" MSG="Idle Timeout > Printer led"
      POWER_OFF_LED
      {% else %}
      RESPOND PREFIX="info" MSG="Idle Timeout > Motors not turned off due to paused print"
    {% endif %}
timeout: 30

########################################
# TMC2208 configuration
########################################

[tmc2209 stepper_x]
uart_pin: PC13
run_current: 0.800
hold_current: 0.400
sense_resistor:  0.11
diag_pin: PB10
stealthchop_threshold: 250
driver_sgthrs = 100

[tmc2209 stepper_y]
uart_pin: PE3
run_current: 0.800
hold_current: 0.400
sense_resistor: 0.11
stealthchop_threshold: 200
diag_pin: PE12
stealthchop_threshold: 250
driver_sgthrs = 100

[tmc2209 stepper_z]
uart_pin: PE1
run_current: 0.800
hold_current: 0.400
diag_pin: PG8
stealthchop_threshold: 250
#driver_sgthrs = 254

[tmc2209 extruder]
uart_pin: PD4
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

#[tmc2208 extruder1]
#uart_pin: PD1
#run_current: 0.800
#hold_current: 0.500
#stealthchop_threshold: 999999

#[tmc2208 extruder2]
#uart_pin: PD6
#run_current: 0.800
#hold_current: 0.500
#stealthchop_threshold: 999999

########################################
# EXP1 / EXP2 (display) pins
########################################

# for the SKR Pro 1
[board_pins]
aliases:
    EXP1_1=PG4, EXP1_3=PD11, EXP1_5=PG2, EXP1_7=PG6, EXP1_9=<GND>,
    EXP1_2=PA8, EXP1_4=PD10, EXP1_6=PG3, EXP1_8=PG7, EXP1_10=<5V>,
    EXP2_1=PB14, EXP2_3=PG10, EXP2_5=PF11, EXP2_7=PF12,  EXP2_9=<GND>,
    EXP2_2=PB13, EXP2_4=PB12, EXP2_6=PB15, EXP2_8=<RST>, EXP2_10=PF13

    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp0"

# See the sample-lcd.cfg file for definitions of common LCD displays.
# https://github.com/KevinOConnor/klipper/blob/master/config/sample-lcd.cfg

[display]
lcd_type: st7920
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_5, ^EXP2_3
click_pin: ^!EXP1_2

# Beeper
[output_pin beeper]
pin: EXP1_1
pwm: True
value: 0
shutdown_value: 0
cycle_time: 0.0024

########################################
# BLTouch Configuration - https://github.com/KevinOConnor/klipper/blob/master/config/sample-probe-as-z-endstop.cfg
########################################

# BLTouch V3 (original) running in open drain mode
# SKR 1.3 reference: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/hardware/SKR-V1.3-PIN.pdf
# BLTouch 3.0 reference: https://docs.wixstatic.com/ugd/f5a1c8_26f2661f0a4542eaacc1586eba8842e7.pdf
# According to this, when the USB is facing towards you, servo pins should be (left to right) colored red (power)-brown (ground)-orange (control pin)
# and the Zmin plug should be (top to bottom) power pin (leave empty), black wire (ground), white wire (sensor pin).

[bltouch]
sensor_pin: ^PA2
control_pin: PA1
stow_on_each_sample: False
samples: 1
speed: 7.5
probe_with_touch_mode: True
sample_retract_dist: 3.0
samples_tolerance: 0.050
samples_tolerance_retries: 2
z_offset = 1.200
y_offset:  0
x_offset: -48

# Z-safe homing that (1) moves the probe up 15mm (2) home XY (3) move to center of bed (4) home Z
# Edit the X100 Y100 for your own center of bed

[safe_z_home]
home_xy_position: 100,100
speed: 120
z_hop: 20
z_hop_speed: 400

########################################
# Bed Mesh and Leveling
########################################

# Example bed_tilt config section
#[bed_tilt]
#points:
#    100,100
#    10,10
#    10,100
#    10,190
#    100,10
#    100,190
#    190,10
#    190,100
#    190,190

# Example bed_mesh config section
[bed_mesh]
speed: 250
horizontal_move_z: 10
mesh_min: 0,30
mesh_max: 150, 150
probe_count: 3,3
algorithm: bicubic
bicubic_tension: .1
relative_reference_index: 0
fade_start: 0.6
fade_end: 10.0
mesh_pps: 2,2

########################################
# Misc
########################################

[virtual_sdcard]
path: /home/pi/gcode_files

# Filament Runout Wired to X-max stop but you can pick any pin you want.
#[filament_switch_sensor filament_sensor]
#pause_on_runout: True
#switch_pin: ^!P1.28

#filament runout sensor
# https://www.reddit.com/r/ender5plus/comments/kb5oe9/klipper_m600_filament_change/
[filament_switch_sensor filament_sensor]
switch_pin: PF8
event_delay: 1.0
pause_on_runout: False
pause_delay: .0001
insert_gcode:
    G4 P300
    LOAD_FILAMENT
runout_gcode:
    M600 X100 Y0 Z50

# This adds pause/resume support
[pause_resume]
recover_velocity: 50
#   When capture/restore is enabled, the speed at which to return to
#   the captured position (in mm/s). Default is 50.0 mm/s.

[display_status]

# This adds the 'respond' G-Code that you can use to send commands back to OctoPrint
[respond]
default_type: echo

# This enables a 'force_move' command ignoring all homing, Z-stops etc. Useful in debugging.
[force_move]
enable_force_move: True

manroop5, 2021-07-20:
This question is for the developer. I think the change in the settings from from "relative to the nozzle" to the "relative to the probe" in bltouch causing these issues.

jakep82, 2021-07-21:
Complicated questions are better discussed on Discourse where you can upload a log. Many people only check into Discord periodically throughout the day, so your post was likely only seen by a few people.  You won't receive help from any developers here.

closed 2021-07-22 by github-actions[bot]