Klipper3d/klipper #4507: TMC2008 drivers not supporting One Wire UART Addressing

ElGranPepo, 2021-07-21:
environment:

- TwoTrees Sapphire Plus printer
- MKS Robin Nano 1.2 board
- 3x2208 (X,Y,Extruder) + 2x2225(Z, Z1) TMC Stepper drivers - using 2208 for 2225 as per comments

Configured in Printer.CFG as **tmc2208** stepper_ with uart_address: being anything other than 0 I get:

 **Option 'uart_address' in section 'tmc2208 stepper_y' must have maximum of 0**

when I change the line to **tmc2209** stepper_ i get no error, one wire is supported on both drivers

klippy.log attached

Watermott TMC datasheet attached.

also, any problems when running these as 2209s? dont want to burn my board 

thank you for such a great software! :)

[klippy.log](https://github.com/KevinOConnor/klipper/files/6855796/klippy.log)

[TMC220x_TMC2224_Datasheet_Rev1.11.pdf](https://github.com/KevinOConnor/klipper/files/6855791/TMC220x_TMC2224_Datasheet_Rev1.11.pdf)

ElGranPepo, 2021-07-21:
this is a bug request

jakep82, 2021-07-21:
Klipper only supports one wire UART on 2209 drivers.

ShohninDmitriy, 2021-07-21:
2208 for 2225 do not support addressing

ElGranPepo, 2021-07-21:
Thanks for the feedback but i cant find this limitation in the documentation. The Trinamic documentation states that its supported

[image]

jakep82, 2021-07-21:
The 2209 supports up to 4 addresses (0-3).  The data sheet you copied from the 2208 clearly states that it supports a single address, and requires analog switches to address more than one IC.  Here's the relevant section from the 2209 data sheet.

```
ADDRESSING MULTIPLE SLAVES
As the TMC2209 uses has a limited number of UART addresses, in principle only up to four ICs can be
accessed per UART interface channel. Adding analog switches allows separated access to more
individual ICs. This scheme is similar to an SPI bus with individual slave select lines (Figure 4.2). With
this scheme, the microstep resolution can be selected via MS1 and MS2 pins (consider actual setting
for addressing).

ElGranPepo, 2021-07-21:
> The 2209 supports up to 4 addresses (0-3). The data sheet you copied from the 2208 clearly states that it supports a single address, and requires analog switches to address more than one IC. Here's [...]

thank you Jake, i understand that we would need to enable analog switching to do this.  it looks like there has been conversation about this in the past:

https://github.com/KevinOConnor/klipper/issues/542

https://github.com/KevinOConnor/klipper/issues/542#issuecomment-414033488

jakep82, 2021-07-21:
Soft UART is already implemented for the 2208, but each driver requires it's own GPIO pin.  It handles bi-directional communication (RX and TX) on a single wire, not multiple drivers.  What you want to do can't be done with your hardware.

ElGranPepo, 2021-07-21:
> Soft UART is already implemented for the 2208, but each driver requires it's own GPIO pin. It handles bi-directional communication (RX and TX) on a single wire, not multiple drivers. What you want to do can't be done with your hardware.

Thank you, Closing.

closed 2021-07-21 by ElGranPepo