Klipper3d/klipper #4511: Updating Pressure Advance tuning guide

Pun-e, 2021-07-22:
Hi there,

Whilst working on printer profiles and fine tuning material profiles etc. I've found the majority of the available documentation good to excellent, if a little hard to navigate. But there's one process that I believe is unnecessarily complicated for something that should be done relatively frequently - tuning your pressure advance.

The current instructions advise downloading the provided stl, some cursory slicer settings guidelines, and the klipper config. Following the current process the generated gCode is thus:
[image]

Versus my proposed gcode:
[image]

Finished print (stopped early) to show that it actually provides useful calibration data
[image]

The current process uses 7.5x as much filament (6g vs 24g) and takes 8x as long (6 min vs 49 min) to provide the _exact same data._

Additionally I must commend the demonstration printer on the excellent quality of their entire extrusion system as the variance in the provided photos is so miniscule that I still don't think that that printer needs any pressure advance.

I would be happy for my prior photo to be used along with some visual markers to help people out, or simply source one of the myriad photos one can find online showing off their much worse (but more educational) pressure advance runs so that first-timers and haven't-done-in-agers can immediately see the limits of what they're looking for without having to turn to reddit, or the discord

I.e. Red arrow = definitively too much advance, Green arrow = definitively not enough advance
[image]

These are my detailed steps on my proposed process:

1. Insert a simple cube into your slicer
2. Scale the X & Y dimensions with respect to the [prusa acceleration calculator](https://blog.prusaprinters.org/calculator_3416/#speed) - for this tutorial I will use a top speed of 100 mm/s, acceleration of 500 mm/s^2, and a final side length of 30mm
3. Make the z height as large as reasonable - for this tutorial we will use a height of 100mm
4. Turn on "Vase mode"
5. Ensure you have disabled all smart acceleration controls - Ensure you have no M204 commands in your start gcode either
6. Ensure that your slicer is maintaining a consistent speed via your gcode preview and is unaffected by any advanced cooling or speed controls
7. Slice!
8. Before printing run the following **two** commands:
9. SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=1 ACCEL=500 - use your previously decided upon acceleration
10. TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.005 - base your advance start on prior testing or leave it at 0 & for the factor simply take a value of 0.3 for direct-drive systems, or 1 for bowden systems and divide by the total height of your gcode - for this example we arrive at a factor of 0.003

The rest of the tuning guide is great.

I'm sorry if this is the wrong avenue for the this kind of "issue" but the guide website is not a wiki, nor is feedback for the guidance website mentioned on your posting guidelines

robthide37, 2021-07-25:
my tool on my website i feel is way better. www.advanced3dprinting.com i modified the marlin tool

KevinOConnor, 2021-07-26:
FWIW, the current documented method has some advantages:
1. Warping can cause subtle problems with reading the desired value.  That's why the current stl file has the "round feet" on it.
2. The inner notches can provide additional visual feedback on the impact of PA.
3. Single wall prints (and similarly vase mode) may work on some printers, but it can produce misleading results on other printers/filaments.  In particular, nozzle movement and uneven cooling can alter the walls.  Also, I've found multiple walls help identify a good PA value because each layer adds to the bulging and makes the impact more noticeable.  So, I would not recommend using single wall prints for this test - at least for new users.
4. The resulting print is sturdy enough that one can keep it around for comparisons with other filaments at a later time.  (Or, for checking if the filament has changed after prolonged storage.)

I should also point out that those wishing to optimize filament or time should be able to modify the START and FACTOR parameters to find/confirm the desired PA while only printing the first cm of the print.

Cheers,
-Kevin

KevinOConnor, 2021-07-26:
> my tool on my website i feel is way better. www.advanced3dprinting.com i modified the marlin tool

FWIW, I found that minor errors in bed level had a significant impact on that style of test.  So, although that test may be faster, I find it less robust.  For general users, I think the big and sturdy "x marks the spot" type of print is a better choice.

Cheers,
-Kevin

blalor, 2021-07-28:
> Warping can cause subtle problems with reading the desired value. That's why the current stl file has the "round feet" on it.

The brim should resolve that.

I use a similar approach as shown here; I found the recommended object too time- (and filament-) consuming for my tastes. I do a 30x30x50mm box with 2 perimeters and no infill or top layer to work well for me. My printer is not as well-tuned as the one in the documentation…   The approach detailed here is also nice because you can print a taller tower in less time while using a smaller tuning factor; I found the recommended factor of 0.02 to be far too high for _my_ bowden setup (Voron v0.0), so a smaller factor and taller object would help novices find a closer PA value with fewer printing iterations.

robthide37, 2021-09-07:
Curious if you still think that 500 accel is best way to check now that we are printing much faster and pa smooth time is sometimes needing to be reduced . I found that 3000 accel and the test on my website let me test every .001 and get insanely accurate results @KevinOConnor  if you could give it a shot with like 50 to 200 mms and 40mm slow 80mm fast lengths 3k or higher accel and tell me what you think. It should only take 10 min and I think the bed issue will be moot point unless mesh is not good

KevinOConnor, 2021-09-27:
As far as I know, 500 accel (and the current PA test in general) works.  It works for me on my printers and with my filaments.  I haven't seen problem reports of grossly inaccurate PA results.

If you've got a different way to calculate PA, then you'll probably want to share that with the Klipper community and gather feedback from many users on lots of different printers.  I'm backlogged on several projects and I'm not really the best person to test these things.

-Kevin

closed 2021-11-09 by github-actions[bot]