Klipper3d/klipper #4519: TUNING_TOWER Command ends before first layer

DeanCording, 2021-07-25:
Trying to run a pressure advance calibration, ran the <TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.005>

The console displayed the following messages:

    16:55:50 Starting tuning test (start=0.000000 factor=0.005000)
    16:56:08 File opened:Makerbot/square_tower.gcode Size:393036
    16:56:08 File selected
    17:02:48 pressure_advance: 0.150000
                   pressure_advance_smooth_time: 0.040000
    17:03:17  Ending tuning test mode
    18:08:34 Done printing file

Looking at the <tuning_tower.py> code at line 59:

      if z < self.last_z - CANCEL_Z_DELTA:
                # Extrude at a lower z height - probably start of new print
                self.end_test()

This cancels the test if the Z height moves down more than CANCEL_Z_DELTA (2mm) on the assumption this is a new print.  However, the START_PRINT macro I use runs a filament purge/prime procedure that involves moving the Z height up and down a few times by more than 2mm, which cancels the TUNING_TOWER test.

This means that I cannot run the TUNING_TOWER command.  If I hadn't noticed the console message saying it the test had been ended, I would not have known it did not work.

I suspect using <SAFE_HOME> with a <z-hop> may cause this issue as well.

A work-around is to place the <TUNING_TOWER> command at the end of the <START_PRINT> macro or after the <START_PRINT>in the GCode file.

DrumClock, 2021-08-04:
HI,
Even if the command:
`TUNING_TOWER COMMAND = SET_PRESSURE_ADVANCE PARAMETER = ADVANCE START = 0 FACTOR = .005`
I place it in the G-code file with the print, just before the actual print of the tower (behind M109, M190 G28), so nothing happens.
The corners of the tower are equally convex along their entire length, no change in pressure advance is visible.

DrumClock, 2021-08-05:
maybe this macro should work ..... I'm testing today

[image]

DrumClock, 2021-08-05:
Hi, now i try it and it works. From each layers, the `pressure_advance` changes.
**Unfortunately I'm at a height of 20 mm from 50 mm and no change of corners .**

bill-orange, 2021-08-10:
I am seeing the same problems.  I have printed multiple tuning towers following the Klipper documentation.  I have not been able to see any change in the printed tower.  Since there are three of us now, I think something is clearly wrong other than operator error.  I have a direct extruder.  At factor 0.010 something should show up.

Immediately before running the tower I ran:

```
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=1 ACCEL=500
TUNING_TOWER COMMAND=SET_PRESSURE_ADVANCE PARAMETER=ADVANCE START=0 FACTOR=.010

```
[printer-20210801_180902.zip](https://github.com/KevinOConnor/klipper/files/6964037/printer-20210801_180902.zip)
[Square Tower Files.zip](https://github.com/KevinOConnor/klipper/files/6964044/Square.Tower.Files.zip)

Thoughts?

keymaster4225, 2021-08-11:
I thought it was just me. Having the exact same issue. Give the tuning tower command but the corners are exactly the whole way up.

bill-orange, 2021-08-11:
Are you using an old school Melzi 8-bit board ( like the early Tronxy or Anets )?  I am wondering if there is an incompatibility between the cheap stepper controllers used on the old 8-bit boards and pressure advance.

keymaster4225, 2021-08-11:
I'm running a tronxy x5sa pro. Runs the chitu v6 board. Has TMC drivers.

bill-orange, 2021-08-12:
I MAY have gotten to most of my problem.  I printed 2 or 3 hollow calibration cubes at high pressure advances.  No change.

I turned off retraction  Voila!  I started seeing the expected behavior on subsequent cubes.  I have not zeroed in on the right value yet.  0.5 is too high and 0.05 is too low.  My hot end was starting to leak, probably complicating things as well.

I will work on this later.  My accelerometer just arrived from China.  @keymaster4225 try running at a high pressure advance with retraction complexly off and see what happens.

keymaster4225, 2021-08-12:
@bill-orange I will try that. I seriously wonder if I left retraction on in my slicer...

Also have an accelerometer coming in a few days.

DrumClock, 2021-08-15:
> I turned off retraction Voila! I started seeing the expected behavior on subsequent cubes. I have not zeroed in on the right value yet. 0.5 is too high and 0.05 is too low. My hot end was starting to leak, probably complicating things as well.

Hi,
in **vase mode**, the slicer will not use retraction ..... so I don't understand why you turned it off

bill-orange, 2021-08-16:
@DrumClock I had to turned retraction off because I had probably turned it on accidently in the past.  In hindsight the hot end leak could have been the bulk of the problem.  A reservoir of molten material could certainly thwart pressure advance.

DeanCording, 2021-09-25:
Reading through the comments on this issue, it seems to have wandered way off track.

The crux of the issue is that Klipper assumes that if the print head moves down (ie Z height decreases) by more than `CANCEL_Z_DELTA` for any reason, then the `TUNING_TOWER` test is ended.  This assumption is not correct in all cases.

KevinOConnor, 2021-09-27:
That is the intended behavior of the TUNING_TOWER command - if there is an extruding move at a particular Z level, then an extruding move at a Z height significantly below that will cancel the mode.

-Kevin

closed 2021-11-09 by github-actions[bot]

bkw777, 2022-02-19:
This got me too. Spent all day trying to figure out why I was following the written directions, and it wasn't working.

Nothing anywhere says "Oh btw if your slicer start gcode has a purge routine that happens to include any Z hop, this will abort the tuning tower and you'll spend the next hour watching a pointless print."

Nor does the console message say that the tuning mode was aborted because it detected Z movement, only that it was aborted.

That is not intuitive behavior. There is no way for the user to even *guess* what's wrong from either the directions or the console messages. Also in my case at least, the Z movement wasn't something I added. Even thinking "I don't know what I did wrong, but if I format and reinstall the whole fluiddpi sd card image, install cura fresh, even in a new user account, surely that will clear it..." wouldn't have cleared it. This is completely obscure hidden secret mystery behavior.

This assumption about Z movement is indeed an unsafe assumption and broken, given what outcome it causes.

A minimal fix that doesn't require a code change, could be just change the documentation to say to enter the tuning tower command in the console after the print starts printing the first layer. You can paste the command in the console ahead of time so that when the time comes all you have to do is press enter or press the send button.