Klipper3d/klipper #4523: tmc_uart change broke the printer for me

ForsakenHarmony, 2021-07-26:
After checking out this commit or anything later, my printer no longer works: 
https://github.com/KevinOConnor/klipper/commit/0075b29081c1d3f8aa001d68ef6c8899cdbd0adc

Message:
```
Unable to read tmc uart 'stepper_b' register IFCNT
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

[klippy.log](https://github.com/KevinOConnor/klipper/files/6880484/klippy.log)

ForsakenHarmony, 2021-07-28:
@KevinOConnor could you tell me why this is not an issue for GitHub?

It has worked previously, the commit I linked breaks on my printer, I don't see a setting to change the UART baud rate in the config

I followed the contact directions and to me, it seems like a "defect in the Klipper software"

KevinOConnor, 2021-07-28:
We only use github to share the results of work done to improve Klipper.

Bug or not, someone needs to diagnose the root cause of your issue.  If you're actively working on diagnosing the issue then you can keep this issue open to relay the results of your work.  If you're looking for help with diagnosing the issue then you should reach out to the Klipper community for assistance.

-Kevin

ForsakenHarmony, 2021-07-28:
I can say that it works fine if I change it to 30k (instead of 40k) manually

KevinOConnor, 2021-07-28:
Okay, thanks.  That's certainly unexpected.  Could you also attach the log from that event.

-Kevin

ForsakenHarmony, 2021-07-28:
The log linked in the original post should contain everything, but here's another one with the newest klipper

[klippy.log](https://github.com/KevinOConnor/klipper/files/6894737/klippy.log)

mdframe, 2021-07-28:
+1 I have the same exact issue with my Rabin Nano v1.3.  How did you solve for this?

ForsakenHarmony, 2021-07-29:
@mdframe you can manually edit the file in the commit linked in the original post and then restart klipper, but be aware that this will prevent you from updating through the various interfaces, you'll have to `git reset HEAD --hard` to be able to update

Alternatively, go into the klipper directory and run `git reset b4fec8502c65db769535b1446db235256cf5b991 --hard` and then restart klipper, this will revert you to before the linked commit, you'll just have to make sure to not update, or it will break again

vladbabii, 2021-07-29:
My setup
1. pi 4 with klipper (latest from yesterday),
2. skr 1.3
3. watterrot 2209 in uart mode 
4. 2 steppers attached to x and y

This is meant to be a development/ gcode test enviroment so it's not a real printer. 

Flow
1. start printer
2. move X or Y 
3. get error in log

> Stats 61956.3: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1703 bytes_read=4893 bytes_retransmit=9 bytes_invalid=0 send_seq=176 receive_seq=176 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99999008  sysload=0.35 cputime=391.174 memavail=1641392 print_time=4.484 buffer_time=0.000 print_stall=0
> Stats 61957.3: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1709 bytes_read=4909 bytes_retransmit=9 bytes_invalid=0 send_seq=177 receive_seq=177 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99998992  sysload=0.35 cputime=391.182 memavail=1641360 print_time=4.484 buffer_time=0.000 print_stall=0
> Stats 61958.3: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1715 bytes_read=4938 bytes_retransmit=9 bytes_invalid=0 send_seq=178 receive_seq=178 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99998973  sysload=0.35 cputime=391.191 memavail=1641376 print_time=4.484 buffer_time=0.000 print_stall=0
> Stats 61959.3: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1721 bytes_read=4954 bytes_retransmit=9 bytes_invalid=0 send_seq=179 receive_seq=179 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99998962  sysload=0.35 cputime=391.198 memavail=1641368 print_time=4.484 buffer_time=0.000 print_stall=0
> Stats 61960.3: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1727 bytes_read=4970 bytes_retransmit=9 bytes_invalid=0 send_seq=180 receive_seq=180 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99998980  sysload=0.32 cputime=391.206 memavail=1641360 print_time=4.484 buffer_time=0.000 print_stall=0
> TMC 'stepper_x' reports GSTAT:      00000001 reset=1(Reset)
> TMC 'stepper_x' reports GSTAT:      00000000
> Stats 61961.3: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=2156 bytes_read=5353 bytes_retransmit=9 bytes_invalid=0 send_seq=202 receive_seq=202 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99998935  sysload=0.32 cputime=391.246 memavail=1641384 print_time=27.852 buffer_time=0.000 print_stall=0
> Stats 61962.3: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=2190 bytes_read=5415 bytes_retransmit=9 bytes_invalid=0 send_seq=205 receive_seq=205 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99998892  sysload=0.32 cputime=391.258 memavail=1641360 print_time=27.852 buffer_time=0.000 print_stall=0
> Stats 61963.3: gcodein=0 mcu: mcu_awake=0.002 mcu_task_avg=0.000008 mcu_task_stddev=0.000009 bytes_write=2224 bytes_read=5491 bytes_retransmit=9 bytes_invalid=0 send_seq=208 receive_seq=208 retransmit_seq=2 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=99998885  sysload=0.32 cputime=391.270 memavail=1641376 print_time=27.852 buffer_time=0.000 print_stall=0

This happens 9 out of 10 starts of the board. I then modified the klipper/klippy/extras/tmc_uart.py file to change both uart values to 9000 then restarted klipper services with systemctl.
After about 20 restarts i see no more issues in log....

Changing back to 40000 baud rate makes these issues return.

@KevinOConnor  can we make this a configurable per [mcu] setting ? Or one per [tmc ...] ? Have a default of 9000 and whoever wants more/less can tweak it for their own use case...

mmhmmh, 2021-07-29:
I can provide a data point. My RAMPS 1.3 board with 4 tmc2209 drivers connected via software UART also has similar issues after a recent update. 

It made the board behave sporadically. For example, when printing, sometimes the "Unable to read tmc uart" comes up. 

Other times, it would make the stepper motors move in a jerky motion and behaves as if the motor wiring has connection issues (I triple-checked the wiring and there is no connection issue). 

This causes significant issues as it basically skews the home position and can result in crashes. 

The issue is very sporadic, sometimes I can do a multi-hour print with no issue but other times it will die when doing basic homing.

Changing the tmc drivers to the a4988 made the issue go away which is not surprising if software UART is causing the issue.

After switching to the spider board, the issue disappeared. I suspect it is because the spider board uses dedicated hardware UART.

ReXT3D, 2021-07-30:
Interesting.

FWIW I conducted **very extensive** amount of UART testing over several weeks on my Duet 3 Mini 5+ with 5 TMC2209 drivers connected with single physical UART bus & using software addressing.  I repeated most of this testing with SKR Mini E3 v2.0 that has four TMC2209 drivers connected with single physical UART.  These two boards worked flawlessly with UART speeds up to and beyond 56k baud with what is effectively the current Klipper code. Older UART code resulted in intermittent UART retransmit issues on both boards.  Kevin suspected all along that there is some kind of timing issue with multiple slave addressing with TMC2209 themselves.

For those interested you can read this discourse thread: [Intermittent TMC UART weirdness](https://klipper.discourse.group/t/intermittent-tmc-uart-weirdness/454)

EddyMI3d, 2021-07-31:
@ForsakenHarmony : Keep in mind that the ports you use for UART must be interrupt capable.

KevinOConnor, 2021-08-01:
There was no change to baud rate on the AVR chips, so this issue is unrelated to any issue on a RAMPS board.

Klipper does not use "hardware gpio interrupts" for any purpose ( https://www.klipper3d.org/FAQ.html#do-i-have-to-wire-my-device-to-a-specific-type-of-micro-controller-pin ) and so is not related to this issue.

@ForsakenHarmony - what board is this on (I can't find any info online related to "mks robin nano v1.3" - just references to "v1.2").  Are the driver uart lines manually wired to the micro-controller, and if so, how?

-Kevin

ForsakenHarmony, 2021-08-03:
@KevinOConnor yes, manually wired, jumper cable from one of the ESP socket pins to the TMC2209s

KevinOConnor, 2021-08-03:
Looks like that wire is glitching during the transition from tx to rx.  That is where I would suggest investigating.

Things to look for - are there any resistors/capacitors on the wire.  Are there any resistors/caps on the board that the wire is attached to.  Is there a solid connection on the wire splice.  Does changing the pin to `^PA5` help.

-Kevin

EddyMI3d, 2021-08-03:
@ForsakenHarmony : What are the pin numbers?

ForsakenHarmony, 2021-08-05:
> Does changing the pin to ^PA5 help.
it does not

> Are there any resistors/caps on the board that the wire is attached to.
looks like there is one to ground, I'll see if there's another pin I can use

[schematic](https://raw.githubusercontent.com/makerbase-mks/MKS-Robin-Nano-V1.X/master/hardware/MKS%20Robin%20Nano%20V1.2_004/MKS%20Robin%20Nano%20V1.2_004%20SCH.pdf) PA5 -> WIFI_CTRL -> WIFI_RST

ForsakenHarmony, 2021-08-05:
Welp, using another pin seems to work, thanks for the help troubleshooting

closed 2021-08-05 by ForsakenHarmony