Klipper3d/klipper #4531: Runaway heating no fault reported

robthide37, 2021-07-27:
Log here 
https://drive.google.com/drive/folders/1DLB4wE2dSBZsuUO0nBkL05AqrDULDKlX?usp=sharing
This has happened 3 times recently and the interface will show the Temps shooting up . It was noticed cause the hotend was smoking and the panel and mainsail interface showed the hotned and bed set to 0 but the Temps are just rising . I assume since klipper does not think the heater is active the verify heater code is not active but this is a scary problem . Please give any thoughts on what cause could be. I don't think it's hardware since both the bed and hotend do it but i propose to keep some type of protection on even when the heater is not active to prevent this

robthide37, 2021-07-27:
So I am realizing the protection is supposed to stay active and it almost seems like it thought it issues shutdown but I don't know I can post the previous log from it happening yesterday here if that helps

jakep82, 2021-07-27:
The link you posted doesn't work, and if Mainsail shows both set to 0, then this is likely a hardware issue.  If the mosfets are stuck closed, Klipper can't do anything to prevent thermal runaway.

I would also suggest you post this on Discourse.  Github isn't used for troubleshooting.

tinymachines3d, 2021-07-27:
@jakep82 the link is public now and contains logs. It only occurs after bed preheating. I have recreated the issue by preheating the bed. As soon as the target is reached, the nozzle and bed start to heat and do not stop with no shut down.

Firmware restart resolves the issue and starting a print does not cause this.

robthide37, 2021-07-27:
> The link you posted doesn't work, and if Mainsail shows both set to 0, then this is likely a hardware issue. If the mosfets are stuck closed, Klipper can't do anything to prevent thermal runaway.
> 
> I would also suggest you post this on Discourse. Github isn't used for troubleshooting.

This is not troubleshooting cause if was stuck mosfet it shouldn't go away with restart and the chances of both a external and internal mosfet going bad at the same time is very low. Also the Temps are still reported the log was 20mb which is why used gdrive. I get that their is alot of troubleshooting posts but when someone has a printer smoking issue it shouldn't be pushed aside so easily

jakep82, 2021-07-27:
From your log:

```
Transition to shutdown state: MCU 'mcu' shutdown: ADC out of range
This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.
```

Klipper did exactly as it should and shut down when your hotend exceeded your max temp of 450.  It then continued heating all the way to 500 even thought the heater was turned off.  From all appearances this is a hardware problem.

jakep82, 2021-07-27:
> Discord is an unorganized place to report issues like this. This is more serious than most.

I agree, that's why I suggested Discourse.

https://klipper.discourse.group/

tinymachines3d, 2021-07-27:
@jakep82 apology thought you meant discord.

tinymachines3d, 2021-07-27:
Log attached forcing the issue stopped with M112. 
[klippy (8).log](https://github.com/KevinOConnor/klipper/files/6888505/klippy.8.log)

KevinOConnor, 2021-07-27:
This requires someone to investigate and find out why it is occurring.  As indicated above, you should post on Discord or Discourse to get help on an investigation.

-Kevin