Klipper3d/klipper #4535: SKR Mini e3 v2.0 +klipper after flash USB not detected

rafikipl, 2021-07-27:
Hi,

I have a similar problem, I have a skr mini e3 v2.0 board, I checked it. normally with Marlin, the USB port is detected, also in octopi, but after uploading the klipper according to the SKR instructions, it is no longer visible, the klipper is prepared according to the recommendations, that is:

>>This file contains common pin mappings for the BIGTREETECH SKR mini
>>E3 v2.0. To use this config, the firmware should be compiled for the
>>STM32F103 with a "28KiB bootloader" and USB communication. Also,
>>select "Enable extra low-level configuration options" and configure
>>"GPIO pins to set at micro-controller startup" to "! PA14".

>>The "make flash" command does not work on the SKR mini E3. Instead,
>>after running "make", copy the generated "out / Klipper.bin" file to a
>>file named "firmware.bin" on an SD card and then restart the SKR
>>mini E3 with that SD card.

after uploading I get:
ls: cannot access '/ dev / serial / by-id / *': No such file or directory

log in the attachment
Someone can help?
[klippy.log](https://github.com/KevinOConnor/klipper/files/6888962/klippy.log)

P.S. from the sticker it looks like the disc is dated 2021.5.19
I can upload Marlin every time and everything works.

closed 2021-07-29 by github-actions[bot]

marknitek, 2021-08-11:
I have the exact same issue... I had everything working on the stock Ender 5 Plus board and now i want to switch to the Mini E3 V2.0

My Board Version is 2021.5.23

I double checked my menuconfig: 
[image]

I reflashed with marlin to see if the usb was bricked or something: **USB is working with marlin**

according to https://github.com/KevinOConnor/klipper/issues/3077#issuecomment-894062505 there are other folks having the same issues with newer  Boards!

blanham, 2021-09-11:
I actually had the same issue on a Pi Zero W, but it went away after I switched to a different USB hub

marknitek, 2021-10-09:
Well i contacted my seller and there is actually a issue with the newer boards having "GD" sticker labels on it. They have another chip on them. I think anyone that bought one of these should be informed from the seller...

I got mine replaced with a real STM board and it works fine now.

Unfortunatly this info is not very accessible it seems to me

ChrisToxz, 2021-10-09:
I just found this announcement, (and of course they announce after they got complains)
https://twitter.com/BigTreeTech/status/1425478829867294722
https://docs.google.com/document/d/1lkGgzv1I1PG7LiQjgpBUTQ7e1_vC5ASGlN6ik3LCNSw/edit